FedeDP/Clight #225: [BUG] -1 when trying to lower DecBl too much

NicolasWebDev, 2021-09-04:
**Clight version the issue has been seen with:**
> 4.7.1

**Used distribution:**
> Archlinux

**Describe the bug**
When the brightness level on my external monitor is at 3% for example, issuing a DecBl 0.05 makes the brightness level go to -1. From there, any IncBl or DecBl stay without effect, because both give a `Failed to validate backlight level request.`.
I have the problem both when using my laptop monitor and my external HDMI monitor.

**Expected behavior**
I would expect the brightness to not go below 0 and to continue responding, even if I issue a DecBl one too many.

```
~$ clight --verbose
Clightd found, version: 5.4.
AC screen backlight curve: y = -0.011678 + 0.060513x + 0.004312x^2
BL
^
|         *
|        *
|
|       *
|      *
|     *
|    *
|   *
|  *
|**
*---------->BR
BATT screen backlight curve: y = -0.011678 + 0.060513x + 0.004312x^2
BL
^
|         *
|        *
|
|       *
|      *
|     *
|    *
|   *
|  *
|**
*---------->BR
init_kbd_backlight(): No such file or directory
Keyboard backlight calibration unsupported.
org.freedesktop.ScreenSaver dbus interface exposed.
Disarmed timerfd on fd 34.
AC cable connected.
Laptop lid opened.
Emitting 'AcState' property
Set timeout of 0s 1ns on fd 37.
Emitting 'LidState' property
Set timeout of 30s 0ns on fd 37.
Set timeout of 0s 1ns on fd 38.
Next alarm due to: Sat Sep  4 05:40:00 2021
Set timeout of 24771s 0ns on fd 38.
Emitting 'Sunrise' property
Emitting 'Sunset' property
Emitting 'DayTime' property
Normal transition to 4000 gamma temp.
Emitting 'NextEvent' property
Emitting 'Temp' property
Set timeout of 0s 1ns on fd 39.
Sensor '/dev/video2' is now available.
Resumed as a sensor is now available.
Emitting 'Temp' property
Emitting 'SensorAvail' property
Captured [5/5] from '/dev/video2'. Ambient brightness: 0.995.
Ambient brightness: 0.995 -> Backlight pct: 1.000.
Set timeout of 120s 0ns on fd 39.
Emitting 'AmbientBr' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.99.
Backlight 'intel_backlight' level updated: 1.00.
Emitting 'Temp' property
Emitting 'BlPct' property
Backlight 'intel_backlight' level updated: 1.00.
Backlight 'GMXK5HA349468' level updated: 1.00.
Emitting 'BlPct' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
et timeout of 30s 0ns on fd 37.
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.95.
Backlight 'intel_backlight' level updated: 0.95.
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.90.
Backlight 'intel_backlight' level updated: 0.90.
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.85.
Backlight 'intel_backlight' level updated: 0.85.
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.80.
Backlight 'intel_backlight' level updated: 0.80.
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.75.
Backlight 'intel_backlight' level updated: 0.75.
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.70.
Backlight 'intel_backlight' level updated: 0.70.
Emitting 'BlPct' property
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.65.
Backlight 'intel_backlight' level updated: 0.65.
Backlight 'GMXK5HA349468' level updated: 0.65.
Backlight 'intel_backlight' level updated: 0.65.
Emitting 'BlPct' property
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.60.
Backlight 'intel_backlight' level updated: 0.60.
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.55.
Backlight 'intel_backlight' level updated: 0.55.
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.50.
Backlight 'intel_backlight' level updated: 0.50.
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.45.
Backlight 'intel_backlight' level updated: 0.45.
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.40.
Backlight 'intel_backlight' level updated: 0.40.
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.35.
Backlight 'intel_backlight' level updated: 0.35.
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.30.
Backlight 'intel_backlight' level updated: 0.30.
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.25.
Backlight 'intel_backlight' level updated: 0.25.
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.20.
Backlight 'intel_backlight' level updated: 0.20.
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.15.
Backlight 'intel_backlight' level updated: 0.15.
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.10.
Backlight 'intel_backlight' level updated: 0.10.
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.05.
Backlight 'intel_backlight' level updated: 0.05.
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: 0.00.
Backlight 'intel_backlight' level updated: 0.00.
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: -1.00.
Emitting 'BlPct' property
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: -1.00.
Emitting 'BlPct' property
Failed to validate backlight level request.
Failed to validate backlight level request.
Failed to validate backlight level request.
Failed to validate backlight level request.
Failed to validate backlight level request.
Failed to validate backlight level request.
Failed to validate backlight level request.
Failed to validate backlight level request.
Set timeout of 30s 0ns on fd 37.
```

Thanks for the great work by the way!

FedeDP, 2021-09-05:
Hi! Thanks for opening this issue!
It seems weird that the issue only happens for secondary screen:
> Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: -1.00.
Emitting 'BlPct' property
Emitting 'BlPct' property
Emitting 'BlPct' property
Backlight 'GMXK5HA349468' level updated: -1.00.

I will give a look to the code and report back!

NicolasWebDev, 2021-09-06:
Hi @FedeDP , actually it happens with both the laptop monitor and with the external monitor, sorry if I wasn't clear.

FedeDP, 2021-09-14:
I will give this a look as soon as I find some time :)

FedeDP, 2021-10-20:
I'll close the issue; let me know if bug happens again (i couldn't reproduce it on latest clight / clightd!)