FedeDP/Clight #226: [BUG]  systemd 234, libelogind not found

promeneur, 2021-09-10:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> 4.7

**Used distribution:**
> ... opensuse

**Describe the bug**

buiding the 4.7 version, i get this error:

-- Checking for one of the modules 'libelogind;libsystemd>=234'
CMake Error at /usr/share/cmake/Modules/FindPkgConfig.cmake:711 (message):
  None of the required 'libelogind;libsystemd>=234' found

systemd version is 246.15

I can't find any libelogind with the software manager

promeneur, 2021-09-10:
same problem with clightd 5.4

-- Checking for one of the modules 'libelogind;libsystemd>=221'
CMake Error at /usr/share/cmake/Modules/FindPkgConfig.cmake:711 (message):
  None of the required 'libelogind;libsystemd>=221' found

FedeDP, 2021-09-10:
Hi! Thanks for opening this issue!
Do you have systemd-devel installed by chance?

promeneur, 2021-09-11:
systemd-devel was the problem.

But now I get a new one :)

[100%] Linking C executable clightd
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld : /usr/lib64/gcc/x86_64-suse-linux/7/../../../../lib64/libddcutil.so : référence indéfinie vers « libusb_set_option »
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/clightd.dir/build.make:560: clightd] Error 1
make[1]: *** [CMakeFiles/Makefile2:113: CMakeFiles/clightd.dir/all] Error 2
make: *** [Makefile:169: all] Error 2

promeneur, 2021-09-11:
ok i found the psolution.

 i must install opensuse packages 
libgusb-devel, typelib-1_0-GUsb-1_0

thanks

FedeDP, 2021-09-11:
Wow great! I will try to update wiki with depa for opensuse!

FedeDP, 2021-09-12:
Wikis have been updated! I am not super sure about deps (each distro has its own naming scheme for libraries/devel packages etc etc; it's a nightmare!), can you check them out?
Thank you!

promeneur, 2021-09-13:
For Clightd

you must add libgusb-devel, typelib-1_0-GUsb-1_0

perhaps typelib-1_0-GUsb-1_0 is a dependency of libgusb-devel

promeneur, 2021-09-13:
If I have time I will create a virtual machine with opensuse leap 15.3. Then i will explore completely what to add for buiding clight and clightd with a default install of opensuse 15.3.

FedeDP, 2021-09-14:
Thank you very much! 
Clightd does not depend upon libgusb, libusb is enough.
I am pretty sure the deps i listed are fine, just the naming may differ (that is an incredible nightmare...)
Thanks anyway!