FedeDP/Clight #227: [BUG] multiuser => clight stop running

promeneur, 2021-09-13:
openSuse 15.3

clight 4.7
clightd 5.4

open a kde session for user1
open a kde session for user2
do different things in user2 session
close user2 session
go to user1 session
read the log of clight

there is no more action by clight after 10 h 33. It is 12 h 45.

see the log
[Clight.log](https://github.com/FedeDP/Clight/files/7153412/Clight.log)

promeneur, 2021-09-13:
the status of clightd

sudo systemctl status clightd
● clightd.service - Bus service to manage various screen related properties (gamma, dpms, backlight)
     Loaded: loaded (/etc/systemd/system/clightd.service; disabled; vendor preset: disabled)
     Active: active (running) since Mon 2021-09-13 09:02:42 CEST; 3h 17min ago
   Main PID: 2776 (clightd)
      Tasks: 3 (limit: 4915)
     CGroup: /system.slice/clightd.service
             └─2776 /usr/libexec/clightd

sept. 13 11:42:54 localhost.localdomain clightd[2776]: [default]|IDLE|: Starting Client 3
sept. 13 11:42:54 localhost.localdomain clightd[2776]: [default]|IDLE|: Starting Client 2
sept. 13 11:42:54 localhost.localdomain clightd[2776]: [default]|IDLE|: Stopping Client 1
sept. 13 11:42:54 localhost.localdomain clightd[2776]: [default]|IDLE|: Stopping Client 0
sept. 13 11:42:54 localhost.localdomain clightd[2776]: [default]|IDLE|: Client 2 -> Idle: 0
sept. 13 11:42:54 localhost.localdomain clightd[2776]: [default]|IDLE|: Client 2 -> Idle: 0
sept. 13 11:42:54 localhost.localdomain clightd[2776]: [default]|IDLE|: Stopping Client 3
sept. 13 11:50:11 localhost.localdomain clightd[2776]: [default]|IDLE|: Removing inotify watSetting Pixel Format: Input/output er>
sept. 13 12:00:16 localhost.localdomain clightd[2776]: Setting Pixel Format: Input/output error
sept. 13 12:10:21 localhost.localdomain clightd[2776]: Setting Pixel Format: Input/output error

FedeDP, 2021-09-14:
Hi! Thanks for opening this issue!
Clight should pause itself and resume when user session becomes inactive/active, when run on a systemd system.
It seems like it is not resuming in your case.
I will give this a look!

FedeDP, 2021-09-15:
You can try to right click on clight desktop file and select "Resume all clight features"; if that's the case (ie: clight does not sees your session back as Active), it should work.

promeneur, 2021-09-17:
i don't find any desktop file !

promeneur, 2021-09-26:
It's a random  phenomenon.

FedeDP, 2021-12-17:
Hi @promeneur ! Any news on this? Ie: has the issue happened again?

promeneur, 2021-12-19:
How to answer with a random phenomenon ?

today i check there is no problem but tomorrow i don't know.

FedeDP, 2021-12-19:
That's unfortunate.
I hoped that perhaps you were able to relate the issue to something.
Thanks anyway!

FedeDP, 2023-03-03:
Since it was like an year ago...is the issue still present? Had you experienced it during the past few months?
Thank you very much for your valuable feedback!

promeneur, 2023-03-03:
No more problem.

closed 2023-03-03 by promeneur