FedeDP/Clight #228: [BUG]

Rashad-707, 2021-09-14:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
5.4 ,installed from aur clight-git

**Used distribution:**
arch

**Describe the bug**
I'm trying using clight to control external monitor brightness with amd gpu, but im getting
```
Clightd found, version: 5.4.
Keyboard backlight calibration unsupported.
Failed to create org.clight.clight dbus interface: File exists
Failed to init. Killing module.
Not a laptop device. Killing UPower module.
Failed to retrieve AC state; fallback to ON_AC and OPEN lid.
Timed out waiting on location provided by Geoclue2. Killing module.
No location provider found; fallback to DAY daytime.
Killing GAMMA as no location provider is available.
```

**Expected behavior**
Brightness change

**To Reproduce**
running clight or any other command give me the same issue.

```
Clightd found, version: 5.4.
AC screen backlight curve: y = -0.024825 + 0.191641x + -0.008928x^2
BL
^
|        **
|      **
|     *
|    *
|   *
|  *
|
| *
|*
|
*---------->BR
BATT screen backlight curve: y = -0.010629 + 0.153844x + -0.007284x^2
BL
^
|
|
|       ***
|     **
|    *
|   *
|  *
| *
|
|*
*---------->BR
Set timeout of 30s 0ns on fd 16.
init_kbd_backlight(): No such file or directory
Keyboard backlight calibration unsupported.
Failed to create org.clight.clight dbus interface: File exists
Failed to init. Killing module.
Disarmed timerfd on fd 35.
Not a laptop device. Killing UPower module.
Failed to retrieve AC state; fallback to ON_AC and OPEN lid.
Set timeout of 0s 1ns on fd 27.
Set timeout of 30s 0ns on fd 27.
Could not find location cache file.
```

FedeDP, 2021-09-14:
Hi! Thanks for opening this issue!
This is indeed a weird issue ;) as always!
Can you share a verbose log? Ie: clight --verbose

FedeDP, 2021-11-07:
Hi, any news on this?
Thanks!

FedeDP, 2021-12-17:
Closing. Feel free to reopen this if you are still encountering the issue!

closed 2021-12-17 by FedeDP