FedeDP/Clight #229: [BUG] sharing webcam => confict with qemu

promeneur, 2021-09-20:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> ...

**Used distribution:**
> ...openSuse Leap 15.3
Clight 4.7

**Describe the bug**

If Clight uses the webcam, then QEMU VM can't access to it.
QEMU can't share a webcam inclusively. It can share only exclusively the webcam.
To solve the conflict, I get a webcam for QEMU and another for the other apps.
But when opening a KDE session, clight gets the QEMU webcam before QEMU. Then QEMU can't get its dedicated webcam.

**Expected behavior**

Clight acquires the webcam dedicated to other apps. For this, we need a setting to choose the webcam, as we found with skype.

**To Reproduce**
Steps to reproduce the behavior.

Please attach a Clight verbose log too!

promeneur, 2021-09-22:
Problem solved, thanks.

It would be better to use the id supplied by "lsusb" command as "046d:08e5" because devname video<x> can change.

FedeDP, 2021-10-07:
> It would be better to use the id supplied by "lsusb" command as "046d:08e5" because devname video can change.

That's an interesting suggestion, never thought of that! Can you open a feature request on clightd about this one?
It is easier to track it ;)
Thank you very much, closing this one in the meantime!

closed 2021-10-07 by FedeDP