FedeDP/Clight #230: [BUG] no Clight item in kde menu

promeneur, 2021-09-20:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> ...4.7

**Used distribution:**
> ...openSuse 15.3

**Describe the bug**
A clear and concise description of what the bug is.

After compiling, I installed clight as root with "make install".

No item relative to Clight is installed in KDE menu.

**Expected behavior**
A clear and concise description of what you expected to happen.

An item is installed in the KDE menu.

**To Reproduce**
Steps to reproduce the behavior.

Please attach a Clight verbose log too!

FedeDP, 2021-09-21:
Hi! thanks for opening this bug report! Can you share output of
`ls /usr/share/applications/clightc.desktop`?

promeneur, 2021-09-21:
@localhost:~> ls /usr/share/applications/clightc.desktop
/usr/share/applications/clightc.desktop
@localhost:~>

promeneur, 2021-09-26:
Hello 

Clight is there again in kde menu !

closed 2021-09-26 by promeneur