FedeDP/Clight #231: [BUG] High cpu usage

samuaz, 2021-10-05:
Hello, thanks for the hard work, I really love this software. For now the only issue that I have is a very high CPU usage 100% under archlinux 5.14.8-arch1-1.

Let me know how I can help to debug this. Thanks

[image]

```
● clightd.service - Bus service to manage various screen related properties (gamma, dpms, backlight)
     Loaded: loaded (/usr/lib/systemd/system/clightd.service; disabled; vendor preset: disabled)
     Active: active (running) since Tue 2021-10-05 01:46:43 -03; 2min 52s ago
   Main PID: 275371 (clightd)
      Tasks: 3 (limit: 38173)
     Memory: 1.4M
        CPU: 2min 29.620s
     CGroup: /system.slice/clightd.service
             └─275371 /usr/lib/clightd/clightd

oct 05 01:46:43 M15R3 systemd[1]: Starting Bus service to manage various screen related properties (gamma, dpms, backlight)...
oct 05 01:46:43 M15R3 systemd[1]: Started Bus service to manage various screen related properties (gamma, dpms, backlight).
```

**this is my config**

```
verbose = true;
resumedelay = 0;
backlight : 
{
  disabled = false;
  restore_on_exit = false;
  no_smooth_transition = false;
  trans_step = 0.05;
  trans_timeout = 30;
  no_auto_calibration = false;
  pause_on_lid_closed = false;
  capture_on_lid_opened = false;
  shutter_threshold = 0.0;
  ac_timeouts = [ 600, 2700, 300 ];
  batt_timeouts = [ 1200, 5400, 600 ];
};
sensor : 
{
  captures = [ 5, 5 ];
  ac_regression_points = [ 0.05, 0.1, 0.1, 0.15, 0.2, 0.2, 0.2, 0.22, 0.25, 0.3, 1.0 ];
  batt_regression_points = [ 0.0, 0.0, 0.0, 0.0, 0.2, 0.3, 0.4, 0.5, 0.6, 0.7, 0.8 ];
};
monitor_override = ( );
keyboard : 
{
  disabled = false;
  dim = false;
  ac_regression_points = [ 1.0, 0.97, 0.93, 0.88, 0.81, 0.74, 0.61, 0.45, 0.29, 0.15, 0.0 ];
  batt_regression_points = [ 0.8, 0.78, 0.75, 0.71, 0.65, 0.59, 0.52, 0.36, 0.23, 0.15, 0.0 ];
  timeouts = [ 15, 5 ];
};
gamma : 
{
  disabled = false;
  restore_on_exit = false;
  no_smooth_transition = false;
  trans_step = 50;
  trans_timeout = 300;
  long_transition = false;
  ambient_gamma = false;
  temp = [ 6500, 4000 ];
};
daytime : 
{
  event_duration = 1800;
  sunrise = "";
  sunset = "";
};
dimmer : 
{
  disabled = false;
  no_smooth_transition = [ false, false ];
  trans_steps = [ 0.05, 0.05 ];
  trans_timeouts = [ 30, 30 ];
  dimmed_pct = 0.2;
  timeouts = [ 45, 20 ];
};
dpms : 
{
  disabled = false;
  timeouts = [ 900, 300 ];
};
screen : 
{
  disabled = false;
  num_samples = 10;
  contrib = 0.1;
  timeouts = [ 30, -1 ];
};
inhibit : 
{
  disabled = false;
  inhibit_docked = false;
  inhibit_pm = false;
  inhibit_bl = false;
};

```

nullobsi, 2021-10-05:
not sure if this still works on latest version, but
```busctl --user set-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf Verbose "b" true```
should enable verbose logging, it would be great if you could try that and upload the resulting log :)

nullobsi, 2021-10-05:
ooh, I see this seems to be an issue with Clightd

it could be related to Clight (and the log may help) but you may want to open an issue on FedeDP/Clightd

samuaz, 2021-10-06:
thanks to response, question after enable the verbose i should use the dbus-monitor to generate the log?

ohh i see, sure I can move the issue to the FedeDP/Clightd

FedeDP, 2021-10-07:
Hi @samuaz  thanks for opening this bug report! 
I will move the bug to clightd repository.
I am currently quite busy with real life; i will give this a look whenever i find some spare time ;) 

Btw it shit a random phenomenon? 
Can you try to attach to running clightd when this happen using strace, and reporting back the output?
```
sudo strace -p `(pidof clightd)` 
```

Thank you very much!
Also, thanks to @nullobsi for your support :)