FedeDP/Clight #232: Lower Brightness Step

fleamour, 2021-10-09:
I would like to use Clight at minimum brightness, something I don't think is engineered yet?
How would I go about stopping Clight from running at startup under GNOME?  I have removed program from GNOME Tweaks startup & killed process but seems to automatically start with each new session.

FedeDP, 2021-10-17:
Hi @fleamour ! And thanks for opening this bug report!
I don't really understand what's the bug you are experiencing, can you be so kind to rephrase it?

> How would I go about stopping Clight from running at startup under GNOME?

That's interesting! Clight installs a global autostart file in `/etc/xdg/autostart/clight.desktop`. To disable it, you need to place a file in `$HOME/.config/autostart/` named clight.desktop copying the global autostart file content, and appending a `Hidden=true` line.
For more information, see here: https://wiki.archlinux.org/title/XDG_Autostart

FedeDP, 2021-11-07:
Hi! Any news on this?
Thanks!

fleamour, 2021-11-10:
I'd like to manually set the brightness cadence/step to a duller setting as way of preference.

FedeDP, 2021-11-10:
Hi! Thanks for reporting back!
Do you mean having a not too bright maximum backlight level?

fleamour, 2021-11-10:
Can this be set?

FedeDP, 2021-11-10:
Yup! That is indeed already done when on battery: max backlight level is 80%.  
You can set the max level by customizing backlight curves regression points: https://github.com/FedeDP/Clight/blob/master/Extra/clight.conf#L105
These points are used to build ambient brightness to screen backlight matching curve on ac and on battery; the first point (0.0) is used as a minimum value while the last one (1.0 or 0.80 in the default config) is used as maximum value.

fleamour, 2021-11-10:
Oh?!? That is really cool. Can I remap points by editing a config file?

FedeDP, 2021-11-10:
Of course you can :)
Rationale is that every webcam/ambient light sensor has quite different brightness points, thus every user is able to customize the curve to suit her/his needs!  
Of course you'll have to restart Clight after editing the config file :) 
Moreover, there is also a gui for Clight: https://github.com/nullobsi/clight-gui, that let you set various settings at runtime using [Clight DBus API](https://github.com/FedeDP/Clight/wiki/Api)!

fleamour, 2021-11-10:
Thanks! I’m on it. Simple enough math I did not see a straight forward way in GUI.

closed 2021-11-10 by fleamour