FedeDP/Clight #233: [BUG] Brightness sensor can't be captured on the Pinephone

coldspark29, 2021-10-17:
I just tried using clight on the Pinephone with zero configuration, but verbose turned on. It can detect the backlight values just fine and also finds a sensor, but then fails to capture it.

Here is the log

```
Clightd found, version: 5.4.
AC screen backlight curve: y = -0.024825 + 0.191641x + -0.008928x^2
BL
^
|        **
|      **  
|     *    
|    *     
|   *      
|  *       
|          
| *        
|*         
|          
*---------->BR
BATT screen backlight curve: y = -0.010629 + 0.153844x + -0.007284x^2
BL
^
|          
|          
|       ***
|     **   
|    *     
|   *      
|  *       
| *        
|          
|*         
*---------->BR
Set timeout of 30s 0ns on fd 15.
Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
Monitoring requests to org.freedesktop.ScreenSaver name owner.
Disarmed timerfd on fd 39.
Not a laptop device. Killing UPower module.
Failed to retrieve AC state; fallback to ON_AC and OPEN lid.
Emitting 'AcState' property
idle_get_client(): Operation not permitted
Clightd idle error.
Failed to init. Killing module.
Set timeout of 0s 1ns on fd 26.
Emitting 'LidState' property
Set timeout of 30s 0ns on fd 26.
53.56 10.05 loaded from cache file.
idle_get_client(): Operation not permitted
Clightd idle error.
Failed to init. Killing module.
Loc distance: 53.56,10.05 -> 91.00,181.00 : 2449.69 km.
New location received: 53.56, 10.05.
Set timeout of 0s 1ns on fd 11.
Emitting 'Location' property
Next alarm due to: Sun Oct 17 17:49:00 2021
Set timeout of 18189s 0ns on fd 11.
Emitting 'Sunrise' property
Emitting 'Sunset' property
Emitting 'DayTime' property
set_temp(): Operation not permitted
Failed to set gamma temperature.
Set timeout of 0s 1ns on fd 12.
Sensor '/dev/iio:device1' is now available.
Resumed as a sensor is now available.
Emitting 'NextEvent' property
Emitting 'SensorAvail' property
capture_frames_brightness(): Operation not permitted
Set timeout of 600s 0ns on fd 12.
^CReceived 2. Leaving.
Latest location stored in cache file!
```

Do I have to add the user to some group maybe? It also says that it is killing uPower. Is that maybe the issue?

FedeDP, 2021-10-17:
Hi @coldspark29 !
I am so proud to hear that you're testing Clight on a pinephone!
Looking at the verbose log you provided, it seems that **Clightd** is always responding EPERM (indeed, not just backlight, but gamma and idle too!).
I am wondering  what's causing the bug. Maybe Clightd is not running as root?
Or polkit is not correctly set up? (the latter one looks the most probable culprit to me!)
To quickly test if it is a polkit issue, you can try:
`busctl call org.clightd.clightd /org/clightd/clightd/Backlight org.clightd.clightd.Backlight GetAll "s" ""`

Ie: if Getters are working, but Setters are not, it means that we have a polkit issue (clightd does not allow non active user sessions to call its Setter methods!)

coldspark29, 2021-10-18:
Yeah, it will be pretty amazing. They have just announced an upgraded version with a faster processor a few days ago, that will have the potential to be usable as daily phone and full fledged PC in your pocket.

FedeDP, 2021-10-19:
Clightd returns EIO when no capture succedeed (ie: none of clight conf.num_captures returned a value !=0).
Try to check dmesg if there is any message about the iio device!
Clightd does not much more than using libudev to read the sensor value btw!

FedeDP, 2021-10-29:
> It really hasn't occured in a while now and I checked that point in the list :)

Yay!!

> Can you make it possible to set the ac_timeout and batt_timeout lower than one second? Since it is the lowest value the transition always occurs in pulses. Using a lower timeout could make it more smooth. Or do you have a better idea?

Mmmh i think i can try! We can try something like: setting the timeout > 100 will mean in ms, below instead seconds.  
I don't want to break existing configurations (by parsing the value in milliseconds by default, instead of seconds).
It would be weird but better than nothing i guess :) (obviously the behavior will be documented in the conf file!)

> I think we will also need the adaptive transition timeout. If you set it too low the brightness changes too fast if you just barely move the phone. If you set it too high, it takes a long time to transition from bright to dark or vice versa.

Mmh so as i suggested before, a fixed transition lenght? eg: i always want the backlight to change in 2s; if it has a large delta, it will make lots of fast backlight change, instead if delta is small, it would make a few slow steps.
What do you think?

FedeDP, 2021-10-29:
A new "timeouts_in_ms" conf option for BACKLIGHT module has been added, that when enabled, treats timeouts as milliseconds values ;) 
In my opinion, this should allow fix other point most probably, as you will have smooth transitions that do not even end before another smooth transition starts. 
Let me know!

coldspark29, 2021-10-30:
Thank you, unfortunately it makes the transitions really slow. :( Does it set all timeouts to ms or only the ac_timeout and batt_timeout?

FedeDP, 2021-10-30:
> Does it set all timeouts to ms or only the ac_timeout and batt_timeout? 

Only ac/batt_timeouts!

> Even when you set the timeouts to 1000 ms, the behavior is different. Something is wrong.

That was a now fixed bug!
Care to try again? Thank you!

coldspark29, 2021-10-30:
And there is one issue that we can not really fix from this side. When the system dims the screen after 30 seconds, clight turns the brightness right back up again. So you have to turn the screen dimming off and let clight handle it. As long as there is no integration into Plasma, people will unlikely use both modules. 

Also when you use clight as a dimmer, the transition back to bright is too slow. Seems like the backlight values are applied to the dimming tool as well. If you then turn off smooth transitions, the automatic brightness control doesn't work anymore.

FedeDP, 2021-10-30:
You can turn clight dimmer off instead!
Dimming tool has its own steps and timeouts for the smooth transition, you may need to tweak them if you wish to use clight dimmer ;)

coldspark29, 2021-10-30:
> That was a now fixed bug!
> Care to try again? Thank you!

Ah no, this did not improve things. Actually even the settings with 1 second from before don't work anymore. The brightness slider is just shuffling back and forth. The brightness is not really decreasing when I cover the sensor with my hand.

coldspark29, 2021-10-30:
> EDT: i now understand what you mean, sorry. In that case yes, you have to use clight dimmer instead of the system one. I cannot think of a better way to handle it, will look into that anyway!

I think we would need some sort of IPC with Plasma here, but that would get messy I think.

FedeDP, 2021-10-30:
> Ah no, this did not improve things. Actually even the settings with 1 second from before don't work anymore. The brightness slider is just shuffling back and forth. The brightness is not really decreasing when I cover the sensor with my hand.

That is weird, when `timeouts_in_ms` is disabled it should behave like before! :( 

> I think we would need some sort of IPC with Plasma here, but that would get messy I think.

Indeed, systemd logind exposes an `IdleHint` property, i need to check whether plasma is using it and add support for it in clight ;)

FedeDP, 2021-10-30:
> Indeed, systemd logind exposes an IdleHint property, i need to check whether plasma is using it and add support for it in clight ;)

I added support for `IdleHint` in Clight: clight will hook IdleHint to manage its value updates: when it is true, clight will manage a dimming state, otherwise it will leave dimming state.  
Moreover Clight will now properly call `SetIdleHint` on logind when it dims/dpms the display.

Note that i don't yet know if kde sets the IdleHint property when it dims the screen; moreover, even if clight correctly handles the logind property, it still tries to set its dimmed backlight level in any case.
There is room for improvements here i guess.
In my opinion, the best option is to let clight manage dimmer too, and disable the builtin feature in plasma.

FInally, `timeouts_in_ms` was wrongly placed under KEYBOARD settings; it is now fixed and moved under BACKLIGHT settings.

coldspark29, 2021-10-30:
Will test it later. Thanks :)

coldspark29, 2021-10-30:
> That is weird, when `timeouts_in_ms` is disabled it should behave like before! :(

It is fixed now :) Unfortunately settings the timeout that low doesn't have the desired effect. Instead of making it smoother it pauses every x ms. I already tried settings the transition timeout to 0, but that doesn't seem to have the effects of no timeout.

FedeDP, 2021-10-30:
What about a couple of seconds {ac,batt} timeouts?
I mean, my android phone reacts to ambient light i'd say with a 1/2s delay; it does not seem to react in real time.

Are you able to record some backlight transitions with various different clight configs? 
That would help me think of a good solution :)
Thanks!

coldspark29, 2021-11-04:
I don't have anything against a timeout of two minute, but the behavior was worse when I tried. I can't test the rest at the moment. The Plasma Mobile shell is broken atm and I have no time to attend to it, because I am currently moving. We will have to delay this till next month.

FedeDP, 2021-11-04:
> I don't have anything against a timeout of two minute, but the behavior was worse when I tried. I can't test the rest at the moment

You mean 2seconds, right? :D 
Btw you have already been super helpful, thank you very much!

> We will have to delay this till next month.

No problem :) I would've released next version with good support for pinephone, but i guess we can wait. We can always wait!

coldspark29, 2021-11-05:
Yeah two seconds... Last thing I remember was that the screen dimming of Plasma was not taken in account, so whatever you tried there did not work. That is the most necessary fix.

FedeDP, 2021-11-06:
Btw is this:
>  Make it possible to set the ac_timeout and batt_timeout to ms

still relevant? I mean, if that does behave bad, i can even drop the feature altogether :) 
I don't like having an option that could possible break non-ALS users, if a 1 second timeout is already fine.

coldspark29, 2021-11-06:
Well, the issue is not the capture timeout, but the pause of adjustment during the capturing. This should be made smoother. But to answer your question, I think that one second is good enough. So you can remove it:

FedeDP, 2021-11-08:
In latest master (1d5ede629ee2c55ecad41103d3fed4c9de89f081)
>  Don't reset the brightness after Plasma has dimmed the screen Kim

should be fixed. Basically, if DIMMER mod is disabled, DISPLAY hooks onto logind IdleHint property and manages that as (un)dimmed message, without touching backlight though (as DIMMER module is disabled).

coldspark29, 2021-11-09:
Sounds great. Will test it when I find the time!

FedeDP, 2021-11-21:
Hi again!  
I am looking to fix the remaining issue; was still wondering if a "fixed_transition_duration" would work fine.
Ie: a milliseconds value that is used as a fixed dimming transition duration. I think that is the right solution; moreover I see lots of use cases for it beside pinephone (i might use it myself too on my laptop!).

IMO the proposed multiplier solution is way too hard to explain and understand, and the outcome would be pretty similar to a fixed transition duration indeed. Am I wrong?

Thanks for your valuable input anyway, i really appreciate that!

FedeDP, 2021-11-21:
In latest master, i implemented a `trans_fixed` option for both BACKLIGHT and DIMMER, that allow users to set a fixed duration for the smooth transitions, given a 1% backlight steps.  
Indeed, it feels so damn smooth (on my laptop at least) that i'm thinking about enabling it by default! What is blocking me is that it would break existing configurations, because enabling the `trans_fixed` will disable other `trans_step` and `trans_timeout` options.

coldspark29, 2021-11-22:
Hi, it's great that you are making more progress. I can't really test it out atm. I am way too busy with renovating my flat and frankly quite overworked. I am very interested though and will test it when I have the time.

FedeDP, 2021-11-22:
Let me make it clear once more: your support is highly appreciated, really!
Thank you very much!

coldspark29, 2021-12-31:
So I just tested again. The issue that clight gets in the way of Plasma's dimming is not solved unfortunately. Have to uncheck it again.

Regarding the trans_timeout, I don't see a difference really. Maybe I haven't configured it right? I am still experiencing stops while it is transitioning.

FedeDP, 2022-01-02:
Hi!
First of all, happy new year! :) 
> The issue that clight gets in the way of Plasma's dimming is not solved unfortunately

I guess Plasma is not setting the IdleHint logind property then :( 

> Regarding the trans_timeout, I don't see a difference really. Maybe I haven't configured it right? I am still experiencing stops while it is transitioning.

Mmh you mean the `trans_fixed` config? If affirmative, and you are still seeing a not-so-smooth transition, most probably the pinephone is not supporting a quick backlight refresh rate (eg: external monitors most of the time behave much worse than internal laptop panels).

coldspark29, 2022-01-02:
Well maybe I configured it wrong. Is it a boolean or do I use it instead of the normal trans_timout?

Is there another possibility regarding the Plasma backlight dimming?

FedeDP, 2022-01-02:
> Well maybe I configured it wrong. Is it a boolean or do I use it instead of the normal trans_timout?

It is a milliseconds duration!

> Is there another possibility regarding the Plasma backlight dimming?

I think that the best thing would be that Plasma actually supports IdleHint. You can see that Clight supports it good enough yourself:
* start Clight in verbose mode without dimmer enabled: `clight --no-dimmer --verbose`
* tail Clight log
* on another terminal, set the systemd logind idlehint: `busctl call org.freedesktop.login1 /org/freedesktop/login1/session/auto org.freedesktop.login1.Session SetIdleHint "b" true`
You will see 
> Entering dimmed state...
Pausing BACKLIGHT: DISPLAY
Pausing SCREEN: DISPLAY
Emitting 'DisplayState' property

printed on Clight log.

(btw thank you! While re-testing this, i found a small bug that i promptly fixed :D )

I can't see any other solution that is DE agnostic. Indeed, it seems weird that Plasma does not set the IdleHint flag already!

FedeDP, 2022-01-22:
@coldspark29  i did some research: gnome is indeed setting the IdleHint property in its session manager: https://gitlab.gnome.org/GNOME/gnome-session/-/blob/main/gnome-session/gsm-systemd.c#L560.

I opened an issue on kde bug tracker: https://bugs.kde.org/show_bug.cgi?id=448942

coldspark29, 2022-01-23:
Hopefully they will do something about it :) Atm I have clight disabled because of this. Like this it just produces more issues than it solves. I also think that you should probably get a phone to test it yourself.

FedeDP, 2022-01-23:
Hi! Can't you disable dimming from KDE and rely only upon clight? I don't remember if there were any issues with that!

coldspark29, 2022-01-23:
Yes, but then it is not dimming at all. I still mainly use my iPhone anyway. The Pinephone is just not ready in many respects, that is why I still hesitate to buy the Pinephone Pro. I think getting clight to behave well on a phone will require a lot of testing under different lighting conditions. The requirements for a phone are just a bit different. That is why I believe that if you want to make clight phone-compatible, it would probably be best to have your own unit to see for yourself. So far only Ubuntu Touch has its own backlight control. I have no idea if an external service for this will be accepted by the OS developers anyway and if our efforts here are really worth it.

FedeDP, 2022-01-23:
> Yes, but then it is not dimming at all.

Interesting; most probably Clightd is not catching "idle" events (because it does not support touch events; that's weird because it looks in `/dev/input/*` though).

> That is why I believe that if you want to make clight phone-compatible, it would probably be best to have your own unit to see for yourself

I agree, but i won't spend my money on something that i won't use daily; i am following Pinephone pro and linux on smartphones in general, and as soon as they are "consumer" ready (ie: i can use them daily without missing much functionalities) i will surely buy one of them.  
For now, i will just wait for some help by the community; oh and btw you are in no way obliged to help me :) you already did it a lot!  
I'll just keep this issue open hoping for more feedbacks! ;)

coldspark29, 2022-01-23:
> Interesting; most probably Clightd is not catching "idle" events (because it does not support touch events; that's weird because it looks in /dev/input/* though).
Guess you misunderstood that. I think the system's dimming feature is useful. There was also an issue that you couldn't turn it back on once deactivated. So I just keep it on for testing purposes.

And don't get me wrong. I will test this further when I feel like having a look at the phone. I just had it back in my hands after two months of being very busy. There was a major performance improvement, but looking at the phone more often than once a month is really not worth it, if you are not actively developing. I just think that we shouldn't duplicate efforts. There will be some brightness control at some point by KDE. They are a big community and good at implementing sophisticated features. You would probably have to change a lot to get clight in that direction and in the end it will not integrate into Plasma neatly. So we shouldn't invest too much effort in this if no one is going to use it in the end.