FedeDP/Clight #236: [BUG] Backlight adjustments reset temperature

Fushko, 2021-11-02:
**Clight version the issue has been seen with:**
> 4.8-92f685c

**Used distribution:**
> Pop OS 21.04

**Describe the bug**
When Clight adjusts the backlight to a new value, the current temperature is lost.

**Expected behavior**
The temperature set should stay the same, disregarding backlight adjustments.

**To Reproduce**
- Start with an early sunset to get the night time temperature:  `clight --sunrise=07:00 --sunset=13:00 --night-temp=3000`
- Let temperature go to 3000
- when Clight detects a difference in ambient light, it changes the backlight as expected, but the temperature resets to 6500

In the logs you can see that at some point I turned off my lamp. That's where I could notice the temperature getting reset, then flicking back to proper value as it was being adjusted. 
If instead I turn off the lamp only after the temperature is done adjusting, it's never set back to the proper value until a restart.

I was also successful in reproducing this issue by changing the backlight manually via the OS control panel. Possibly a distro/GNOME issue?

```
Config file /home/fushko/.config/clight.conf not found.
Clightd found, version: 5.5-c78f9c1.
AC screen backlight curve: y = -0.024825 + 0.191641x + -0.008928x^2
BL
^
|        **
|      **  
|     *    
|    *     
|   *      
|  *       
|          
| *        
|*         
|          
*---------->BR
BATT screen backlight curve: y = -0.010629 + 0.153844x + -0.007284x^2
BL
^
|          
|          
|       ***
|     **   
|    *     
|   *      
|  *       
| *        
|          
|*         
*---------->BR
init_kbd_backlight(): No such file or directory
Keyboard backlight calibration unsupported.
Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
Monitoring requests to org.freedesktop.ScreenSaver name owner.
Disarmed timerfd on fd 36.
AC cable disconnected.
Laptop lid opened.
Emitting 'AcState' property
Disarmed timerfd on fd 39.
Emitting 'LidState' property
Set timeout of 0s 1ns on fd 40.
Next alarm due to: Wed Nov  3 06:30:00 2021
Set timeout of 49419s 0ns on fd 40.
Emitting 'Sunrise' property
Emitting 'Sunset' property
Emitting 'DayTime' property
Normal transition to 2500 gamma temp.
Emitting 'NextEvent' property
Emitting 'Temp' property
Set timeout of 0s 1ns on fd 41.
Sensor '/dev/iio:device0' is now available.
Resumed as a sensor is now available.
Emitting 'Temp' property
Emitting 'SensorAvail' property
Captured [5/5] from '/dev/iio:device0'. Ambient brightness: 0.181.
Ambient brightness: 0.181 -> Backlight pct: 0.243.
Set timeout of 5s 0ns on fd 41.
Emitting 'AmbientBr' property
Emitting 'BlPct' property
Backlight 'amdgpu_bl0' level updated: 0.33.
Emitting 'BlPct' property
Backlight 'amdgpu_bl0' level updated: 0.28.
Emitting 'BlPct' property
Backlight 'amdgpu_bl0' level updated: 0.24.
Emitting 'BlPct' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Captured [5/5] from '/dev/iio:device0'. Ambient brightness: 0.181.
Ambient brightness: 0.181 -> Backlight pct: 0.243.
Set timeout of 5s 0ns on fd 41.
Emitting 'AmbientBr' property
Emitting 'BlPct' property
Backlight 'amdgpu_bl0' level updated: 0.24.
Emitting 'BlPct' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Captured [5/5] from '/dev/iio:device0'. Ambient brightness: 0.181.
Ambient brightness: 0.181 -> Backlight pct: 0.243.
Set timeout of 5s 0ns on fd 41.
Emitting 'AmbientBr' property
Emitting 'BlPct' property
Backlight 'amdgpu_bl0' level updated: 0.24.
Emitting 'BlPct' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Captured [5/5] from '/dev/iio:device0'. Ambient brightness: 0.181.
Ambient brightness: 0.181 -> Backlight pct: 0.243.
Set timeout of 5s 0ns on fd 41.
Emitting 'AmbientBr' property
Emitting 'BlPct' property
Backlight 'amdgpu_bl0' level updated: 0.24.
Emitting 'BlPct' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Captured [5/5] from '/dev/iio:device0'. Ambient brightness: 0.000.
Ambient brightness: 0.000 -> Backlight pct: 0.000.
Set timeout of 5s 0ns on fd 41.
Emitting 'AmbientBr' property
Emitting 'BlPct' property
Backlight 'amdgpu_bl0' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Captured [5/5] from '/dev/iio:device0'. Ambient brightness: 0.000.
Ambient brightness: 0.000 -> Backlight pct: 0.000.
Set timeout of 5s 0ns on fd 41.
Emitting 'AmbientBr' property
Emitting 'BlPct' property
Backlight 'amdgpu_bl0' level updated: 0.14.
Emitting 'BlPct' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Emitting 'Temp' property
Captured [5/5] from '/dev/iio:device0'. Ambient brightness: 0.000.
Ambient brightness: 0.000 -> Backlight pct: 0.000.
Set timeout of 5s 0ns on fd 41.
Emitting 'AmbientBr' property
Emitting 'BlPct' property
Backlight 'amdgpu_bl0' level updated: 0.09.
Emitting 'BlPct' property
Captured [5/5] from '/dev/iio:device0'. Ambient brightness: 0.181.
Ambient brightness: 0.181 -> Backlight pct: 0.243.
Set timeout of 5s 0ns on fd 41.
Emitting 'AmbientBr' property
Emitting 'BlPct' property
Backlight 'amdgpu_bl0' level updated: 0.14.
Emitting 'BlPct' property
Backlight 'amdgpu_bl0' level updated: 0.19.
Emitting 'BlPct' property
Backlight 'amdgpu_bl0' level updated: 0.24.
Emitting 'BlPct' property
Backlight 'amdgpu_bl0' level updated: 0.24.
Emitting 'BlPct' property
^CReceived 2. Leaving
```

FedeDP, 2021-11-02:
Hi @Fushko ! Thanks for opening this issue!
From the provided log file, i can exclude that Clight is doing any wrong work for sure; i can see just one
`Normal transition to 2500 gamma temp.`
log, thus clight is not touching gamma anymore.
Are you on wayland? I assume no, because gamma would not work on wayland under gnome (the required protocol is only implemented by wlroots compositor as far as i know).
It is a weird behaviour anyway, but we will understand the root cause!

Fushko, 2021-11-02:
Hi, thanks for the response!
Yep, I'm on X11.

FedeDP, 2021-11-02:
Interesting! So, without clight and using sysfs, same bug happens?
Maybe a bug in your graphic drivers?

Fushko, 2021-11-02:
> Interesting! So, without clight and using sysfs, same bug happens?

Yes, it appears so. Very weird. 

> Maybe a bug in your graphic drivers?

I'm on a laptop I bought recently with AMD vega 10, so I'm just using the kernel drivers.

`uname -r` prints `5.13.0-7620-generic`, if that helps.

FedeDP, 2021-11-02:
Love the behavior :joy:
Can you try in a non X env? Always through sysfs i mean!

FedeDP, 2021-11-02:
Also, I saw you are the dev of gammy! Seems very nice, even if i still haven't tried it! :)

Fushko, 2021-11-02:
> Can you try in a non X env? Always through sysfs i mean!

Awesome! I switched tty after setting low temp, then, changing backlight doesn't cause a reset anymore. Still not sure what the cause is though.

> Also, I saw you are the dev of gammy! Seems very nice, even if i haven't tried it yet! :)

Thanks, but it's pretty abandoned and nowhere near as good as your app :)

FedeDP, 2021-11-02:
It seems like either gnome is messing or X is. Can you try with kde or xfce?

Fushko, 2021-11-03:
I've tried ubuntu 21.10 with gnome and X11: the issue did not occur.
I'd definitely exclude X11 (this issue would have popped up sooner otherwise), so I'm thinking it's related to Pop Os specifically. 
Admittedly, gnome versions are different - 40.4 on Ubuntu and 3.38.5 on Pop - I doubt this is relevant though.

FedeDP, 2021-11-03:
Super interesting! Maybe popOS has some program in the background that does the trick?
The issue matches perfectly the behavior you're experiencing...can you double check your gnome startup applications under popOS?

Fushko, 2021-11-03:
Found the problem! I just realized I could look at the process that started using CPU as I moved the backlight slider... it was the "colord" daemon. Disabled it and the problem is gone :D Thanks for the help!

FedeDP, 2021-11-04:
That's great to hear!
You're welcome sir! :) I'll leave this open for now to remember to add a note to the FAQs!

FedeDP, 2021-11-06:
I added the new entry in the FAQs!
I think we can consider this as closed! Thank you very much!

closed 2021-11-06 by FedeDP