FedeDP/Clight #237: [FEATURE REQ] add sunrise and sunset offsets

elypter2, 2021-11-05:
**Is your feature request related to a problem? Please describe.**
 the indoor lighting conditions do not always match up with natural sunlight or some people want the day on the display appear longer or shorter than the the actual daylight

**Describe the solution you'd like**
an offset would allow you to still depend on the sunlight but deviate from it by a custom amount.

**Describe alternatives you've considered**
a sunrise and sunset offset that can have a positive or negative amount of minutes as values.

FedeDP, 2021-11-06:
Hi! That's a neat idea, thanks for the suggestion!
It is a very simple implementation yet I think it can be useful.
I will implement this :)

FedeDP, 2021-11-07:
Hi! This is now implemented in master :)

commit bb66db4e (2021-11-07): Add support for sunrise and sunset events offset, to allow for even finer granularity. Fixes #237.
closed 2021-11-07 by FedeDP