FedeDP/Clight #240: [BUG] Clightd (idle) error

danjenson, 2021-12-15:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> 4.7

**Used distribution:**
> Void Linux

**Describe the bug**
Clight is erroring out.

**Expected behavior**
It would update backlight appropriately.

**To Reproduce**
1. Install `Clight` from VOID Linux packages

Please attach a Clight verbose log too!
``
   1 (W)[23:53:57]{config.c:400} Config file /home/danj/.config/clight.conf not found.
   2
   3 Clight
   4 * Software version:     4.7
   5 * Global config dir:        /etc/default
   6 * Global data dir:      /usr/share/clight
   7 * Starting time:        Tue Dec 14 23:53:57 2021
   8
   9 Starting options:
  10
  11 ### GENERIC ###
  12 * Verbose (debug):      Disabled
  13 * ResumeDelay:      0
  14
  15 ### BACKLIGHT ###
  16 * Smooth trans:     Enabled
  17 * Smooth steps:     0.05
  18 * Smooth timeout:       30
  19 * Daily timeouts:       AC 600  BATT 1200
  20 * Nightly timeout:      AC 2700 BATT 5400
  21 * Event timeouts:       AC 300  BATT 600
  22 * Backlight path:       Unset
  23 * Shutter threshold:        0.00
  24 * Autocalibration:      Enabled
  25 * Pause on lid closed:      Disabled
  26 * Capture on lid opened:        Disabled
  27 * Restore On Exit:      Disabled
  28
  29 ### SENSOR ###
  30 * Captures:     AC 5    BATT 5
  31 * Device:       Unset
  32 * Settings:     Unset
  33
  34 ### KEYBOARD ###
  35 * Dim:      Disabled
  36 * Timeouts:     AC 15   BATT 5`
 57 * Smooth timeout:       ENTER: 30, EXIT: 30
  58 * Timeouts:     AC 45   BATT 20
  59 * Backlight pct:        0.20
  60
  61 ### DPMS ###
  62 * Timeouts:     AC 900  BATT 300
  63
  64 ### SCREEN ###
  65 * Timeouts:     AC 30   BATT -1
  66 * Contrib:      0.10
  67 * Samples:      10
  68
  69 ### INHIBIT ###
  70 * Docked:       Disabled
  71 * PowerManagement:      Disabled
  72 * Backlight:        Disabled
  73
  74 (I)[23:53:57]{main.c:127}   Clightd found, version: 5.4.
00:37 danj@m2 ~
  76 (I)[23:53:57]{interface.c:180}  org.freedesktop.ScreenSaver dbus interface exposed.
  77 (I)[23:53:57]{upower.c:54}  AC cable connected.
  78 (I)[23:53:57]{upower.c:65}  Laptop lid opened.
  79 (W)[23:53:57]{idler.c:22}   Clightd idle error.
  80 (W)[23:53:57]{dimmer.c:55}  Failed to init. Killing module.
  81 (W)[23:53:57]{idler.c:22}   Clightd idle error.
  82 (W)[23:53:57]{dpms.c:48}    Failed to init. Killing module.
  83 (W)[23:54:27]{location.c:89}    Timed out waiting on location provided by Geoclue2. Killing module.
  84 (W)[23:54:27]{daytime.c:68} No location provider found; fallback to DAY daytime.
  85 (W)[23:54:27]{gamma.c:91}   Killing GAMMA as no location provider is available.
```

FedeDP, 2021-12-17:
Hi! Thanks for opening this issue!

It seems kinda weird that Clight is failing to init idle monitors for dimmer and dpms; the error comes from here: https://github.com/FedeDP/Clight/blob/master/src/utils/idler.c#L23
Can you share `journalctl -b -u clightd.service` output, when this happens?
Thanks!

danjenson, 2021-12-18:
Void linux actually uses `runit` instead of `systemd`, so I don't have those logs.

danjenson, 2021-12-19:
This works if I run `sudo dbus-run-session clight`. If I simply run, `dbus-run-session clight`, it returns that idle error. So, it seems this is a permissions issue. Are there certain groups that provide the proper authorization?

FedeDP, 2021-12-19:
Nice catch! On which DE are you on?

danjenson, 2021-12-19:
DE? I am on Void Linux? I'm not sure what DE means lol

FedeDP, 2021-12-19:
I meant Desktop Environment! :)

danjenson, 2021-12-19:
[image]

FedeDP, 2021-12-19:
Thanks! How do you start bso
wm?
Can you try:
exec dbus-launch --exit-with-x11 bspwm

To start a dbus session with all the correct dbus env set, within your session?

danjenson, 2021-12-19:
I did that, and then I have `dbus-run-session clight` in my bspwmrc, but it still requires sudo to function.

FedeDP, 2021-12-19:
:( unfortunate.
Btw Clight does not require any root privilege, that's the point of having a root backend service (Clightd).  
I am sorry i cannot help you much, i do not know bspwm; quick searches revealed the dbus-launch workaround i posted before, but if that does not work without sudo, i fear i cannot help you more.

Did you try to start bspwm with dbus-launch and then starting clight program without any dbus-run-session too?

FedeDP, 2021-12-19:
Uh moreover Clight requires an Active polkit session!

danjenson, 2021-12-20:
I have done both of those things, to no avail. I have produced an issue on the void package repo:
https://github.com/void-linux/void-packages/issues/34630

danjenson, 2021-12-20:
This polkit rule solved it:
```
cat /etc/polkit-1/rules.d/80-clightd.rules
/* Allow any user to call clightd without authentication */
polkit.addRule(function(action, subject) {
    if (action.id.indexOf("org.clightd.clightd.") == 0) {
        return polkit.Result.YES;
    }
});
```

closed 2021-12-20 by danjenson

FedeDP, 2021-12-20:
Great news, happy to hear you solved your issue!
There is a note on the Clightd wiki about polkit (i don't know why i forgot to link it to you): https://github.com/FedeDP/Clightd/wiki/Polkit-Support