FedeDP/Clight #241: [FEATURE REQ] Switch between internal camera and webcam when docked

danjenson, 2022-01-11:
When I use my laptop stand alone, I'd like it to use the internal webcam to set the backlight, but when I dock it, I'd like it to use the webcam plugged into the dock. Unfortunately, it seems to continue trying to use the internal webcam (which is bad because the laptop is closed -- this sometimes leads to turning the brightness of one of my monitors to 0). 

**Describe the solution you'd like**
I'd like it to be able to try all external webcams before internal webcams, or allow me to specify two devices in `devname` in the order I want them to be tried.

**Describe alternatives you've considered**
I permanently set `devname` to `video4` while I'm docked. But this will only work while docked..

FedeDP, 2022-01-16:
Hi! Thanks for opening this feature request!
Indeed you are right, i myself ran into this issue sometimes; I think that a proper solution, as you suggested, would have to define a list of devices to be used, each with its set of settings and possibly different regression points.  

This is not trivial after all, but totally feasible. 
I am currently planning to release a new Clight version soon:tm:.  
I think this one could be the big thing for the release after that (ie: possibly the 4.8/5.0 release), together with better Pinephone support.  

Another thing that came in my mind lately was to allow to set different curves/settings for each supported sensor type, to allow for more flexibility.  
I think those 2 features would be strictly related and i need to think a bit about the actual best way to implement this.

Thanks!

danjenson, 2022-01-28:
Thank you for all the great work!

FedeDP, 2023-06-07:
Hi! Thanks for your suggestions!
The former adds yet another config key and I'd avoid it :)
The latter is indeed a simple and clean solution but it would cost quite a bit of lost seconds in useless captures (ie: imagine you got 3 webcams, 2 dark one bright, you need to capture 5 frame from each of them, costing around 3-4s).
It's not a big deal though, it could actually make sense!
Thanks again for the input!

Otherwise, i would love to understand whether udev exposes the "camera exists and is internal but lid is closed" information (i don't really think so though...). If that was true, it would just be a matter of implementing the check in Clightd camera selection algorithm (that today just uses the first found).

FedeDP, 2023-06-08:
@supermarin @danjenson https://github.com/FedeDP/Clightd/pull/101 should improve the situation: Clightd will now prefer camera devices with higher sysnum (ie: /dev/video2 over /dev/video0).
This means that when an external camera is attached, Clightd will use it, else it will use the internal one.
What do you think? It is a simple fix yet i think it will solve most of the issues!

supermarin, 2023-06-08:
@FedeDP on paper that sounds pretty genius! will test once an update is released and report back.

FedeDP, 2023-06-08:
Merged in Clightd master if you want to try :)

supermarin, 2023-06-22:
Finally got to give it a shot: seems like hot-plugging doesn't work, i.e. once started, the laptop's camera is cached as the camera to use. Restarting clight systemd service fixes the problem

FedeDP, 2023-06-23:
> the laptop's camera is cached as the camera to use.

Mmmh that is weird since Clightd does not cache anything. Are you by chance using pipewire instead of camera sensor?
To quickly test the behavior, you can trigger a capture by issuing:
```
busctl call org.clightd.clightd /org/clightd/clightd/Sensor/Camera org.clightd.clightd.Sensor Capture "sis" "" 5 ""
```
This will enforce Camera sensor. Unfortunately, if you are using Clightd built with pipewire, pipewire has higher priority even if it is experimental (that's because pipewire allows clightd to capture frames while other applications, using pipewire, are already using the webcam).
But in pipewire i've yet to understand how the logic used for Camera can be applied :)

FedeDP, 2023-06-23:
Uh i was able to reproduce the issue :/ Will work on it asap!

FedeDP, 2023-06-24:
https://github.com/FedeDP/Clightd/pull/104 should fix the issue.
Previously, i took for granted that an higher sysnum would be used for external webcams (ie that `/dev/video2` is always external while `/dev/video0` is always the internal one).
It does not seem to be true though; i am now relying on udev `USEC_INITIALIZED` property that exposes the time the devices was initialized by udev and seems the correct approach. 
I just merged the PR and the fix is now available on Clightd master; let me know!
Moreover, i also ported Pipewire sensor to use same logic as Camera sensor.

supermarin, 2023-06-26:
Dang, good job @FedeDP! I learned something from this.
Will be able to try and report back later this week. If you don't mind cutting a patch release, that'd be appreciated.

FedeDP, 2024-03-09:
I will close this issue since it should be solved!

closed 2024-03-09 by FedeDP