FedeDP/Clight #242: [FEATURE REQ] Automatic brightness adjust based on content

ArijitKHaldar, 2022-01-23:
Your project is pretty interesting..but I felt that something can be added to make it even better. Would you like to take a look at [Lumen](https://github.com/anishathalye/lumen) and see if we can get that behaviour in Clight?

FedeDP, 2022-01-23:
Hi! Thanks for opening this feature request!
I think it can be surely done and it is a great idea!  
I envision it like this:
* `contentBased` conf option under Backlight settings to enable this mode
* when it is enabled, Backlight module will capture ambient brightess like it normally does but won't set screen backlight
* instead, Screen module will compute current screen brightness (just like it does now), **and** will set screen backlight based on a mixture of current ambient brightness and screen brightness, with an algorithm that uses ambient brightness as a 'sliding window' and screen brightness to decide the correct point inside the sliding window.

For example:
* ambient brightness -> 0.78.
* Sliding window -> 0.5 - 1.0
* screen brightness -> 0.2
* final backlight value -> 0.9 (ie: 1.0 - (0.2 * 0.5))

Moreover, Clight would expose new dbus API to let Screen module do a capture, so that one can manually trigger the behavior.
Thing is, lowering Screen timeout too much can have a huge impact on performance while on battery. Therefore i think that on battery the behavior should be mostly manual; instead when connected to AC, the behavior can be fully automatic (with eg: a screen capture every 2s).
WDYT?

Again, thank you very much for your input!

ArijitKHaldar, 2022-01-29:
Glad to be helpful in whatever small way possible...

That's a great idea for determining required brightness based on sliding window algorithm. Would love to try out your implementation of the idea.
Btw, may I request for some user-side control such as calling with manual flags when starting Clight for using on machines where camera might be an external one and used in a very different lighting condition (for example taking successive images of some objects for collecting data for ML training data creation).

What I mean to say is, Clight is already using the required modules for screen brightness control, so... personally, I prefer to use one program for a specific task on all my machines to make things simpler for me. Then, I can override any one of the functions manually in a simple script, if using on a machine that may not benefit from the ambient lighting, but will benefit from percentage of screen content that is white...or of brighter tones...**or** maybe the other way round (probably something like --no-camera or --no-screen-content)

So, it would be something like this:
- Default mode (what Clight presently is doing)
- Screen Content mode (your suggestion to my feature request)
- Manual mode
    - with all features off, Clight would be a simple brightness changing software based on user keyboard shortcuts)
    - with screen content (automatic brightness adjustment based on screen content)
    - with ambient light sensor/camera
    - screen content+light sensor/camera

FedeDP, 2022-01-30:
Hi!
On PR https://github.com/FedeDP/Clight/pull/244 (`feat_content_based` branch) you can find the implementation of the new screen content feature, that is indeed a full reworking on the old screen backlight compensation tool.  
You can try it out, if you don't mind! Any feedback is appreaciated ;)  

> Default mode (what Clight presently is doing)

Present!
> Screen Content mode (your suggestion to my feature request)

Present!
> with all features off, Clight would be a simple brightness changing software based on user keyboard shortcuts) 

You can enable `backlight.no_auto_calibration` (https://github.com/FedeDP/Clight/blob/master/Extra/clight.conf#L89) for that!

> with screen content (automatic brightness adjustment based on screen content) 

This is not so useful IMHO, without knowing ambient brightness (ie: using it together with backlight captures)

> with ambient light sensor/camera 

Without auto calibration enabled and with screen-content based **disabled** -> you can have a keyboard shortcut to trigger the "Capture" action (same that you can find on Clight desktop file)

> screen content+light sensor/camera 

Same as above but with screen-content based **enabled**

To trigger Clight Capture, you can use Clight DBus API:
`busctl --user call org.clight.clight /org/clight/clight org.clight.clight Capture "bb" false false`

or use the provided desktop file!

FedeDP, 2022-09-24:
Just wanted to let you know that the feature is now in master, if you are still looking forward to play with it :D

ArijitKHaldar, 2022-09-26:
It's such an honor...being updated about this after so many months directly from you !! Thank you sir...I really appreciate it. Btw, I was using acpilight for the time being, since I was busy with my dissertation work. But now its done, so, I'll definitely check it out.

Actually things didn't work very well when I last tried it around March. I was having some issues integrating it with my XMonad WM. Had some high cpu usage issues for me. But, I'll try it again this time. 😄

FedeDP, 2023-03-03:
Since this is a released feature (even if not perfect perhaps but pretty tweakable) I'll close this issue :)
Thank you!

closed 2023-03-03 by FedeDP