FedeDP/Clight #243: [BUG] clight: error while loading shared libraries: libgsl.so.25: cannot open shared object file: No such file or directory

coldspark29, 2022-01-29:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> 409b255cf2608388bb0d794b7ad8b746b37acf72

**Used distribution:**
> Manjaro

**Describe the bug**
clight can't be executed. Error message
clight: error while loading shared libraries: libgsl.so.25: cannot open shared object file: No such file or directory

FedeDP, 2022-01-29:
Hi! Most probably you upgraded gsl and now Clight needs to be built again. Can you try to re-install Clight?

coldspark29, 2022-01-29:
I already tried that. Doesn't help. Weird.

FedeDP, 2022-01-30:
Mmh what's the output of: 
ls /use/lib/libgsl*
?

coldspark29, 2022-01-30:
> ls /usr/lib/libgsl*

It's .so.27 and not .so.25. It should be automatically rebuilt with that, shouldn't it?

/usr/lib/libgsl.so /usr/lib/libgsl.so.27 /usr/lib/libgsl.so.27.0.0 /usr/lib/libgslcblas.so /usr/lib/libgslcblas.so.0 /usr/lib/libgslcblas.so.0.0.0

FedeDP, 2022-01-30:
Yep, that's weird! Perhaps Clight is being built with a cached makefile that links to old so.25?
What about:
pkg-config --libs gsl
?

coldspark29, 2022-01-30:
Ah I got it. There was still the old version installed in /usr/local, which I manually compiled. Thanks for the help :)

closed 2022-01-30 by coldspark29