FedeDP/Clight #245: [question ] devname = "<webcam>";

promeneur, 2022-04-09:
In the past, I suggested that it would be handy to use what "lsusb" uses to name the webcam because today it is difficult to know the device system name. For example, with openSUSE curiously my webcam device name is "video2"and moreover if I unplug then plug my webcam to another USB connector, the device name changes.

- "046d:08e5"
or better user-friendly
-  "Logitech, Inc. HD Pro Webcam C920".

You answered that it's a good idea.

Is this feature supplied with the last version ?

FedeDP, 2022-04-09:
Hi! I've yet to implenent the logic (that will go in Clightd btw :) ).
Currently i am quite busy with my job and with real life but this is a really simple feature; sorry if I forgot about it!
I will try to deliver it in let's say a couple of weeks!

FedeDP, 2022-04-09:
Hi! Latest Clightd master supports specifying ATTR{name} for udev devices, ie: this string:
```
$ udevadm info -n /dev/video0 -a | grep ATTR{name}
ATTR{name}=="HD User Facing: HD User Facing"
```
Care to test? :)

promeneur, 2022-04-10:
I am not sure what to do ?

FedeDP, 2022-04-10:
Hi! You should use the command in the other message (eventually changing /dev/videoX) to discover your webcam human readable name attribute. Then use that string as devname in Clight!

promeneur, 2022-04-10:
I use

devname = "HD Pro Webcam C920";

This work perfectly with master branch of clightd.

Thanks

promeneur, 2022-04-10:
With "lsusb" you don't need to know the system name of your camera as "video2" for example to find the friendly name of it.

With "lsusb" I get

Bus 001 Device 006: ID 046d:08e5 Logitech, Inc. HD Pro Webcam C920

The first par t of the text "Logitech, Inc." is the trademark.

The second part of the text "HD Pro Webcam C920" is the model (the friendly name".

FedeDP, 2022-04-25:
Hi! In my case `lsusb` output differs a bit from the `udevadm info -n /dev/video0 -a | grep ATTR{name}` output :) 
Btw i need to add a note in the config file!

closed 2022-04-25 by FedeDP