FedeDP/Clight #247: [FEATURE REQ] Device support: Samsung Galaxy Book Flex2

RubenKelevra, 2022-04-19:
**Is your feature request related to a problem? Please describe.**
My Samsung device got a brightness sensor – which is under Windows only, used to turn on/off the keyboard backlights (which can be adjusted in brightness or turned off completely).

Neither the brightness sensor nor the keyboard backlights (nor the secondary camera of the device) is showing up under Linux – which is kind of strange.

The 720p webcam above the display is working **fine** for clight, so this ticket isn't about that I'm unable to use clight - it works great! :)

I noticed that the 720p device show's up with 3 possible input devices, which is kind of weird to me:

```console
$ v4l2-ctl --list-devices
720p HD Camera: 720p HD Camera (usb-0000:00:14.0-6):
        /dev/video0
        /dev/video1
        /dev/media0
$ v4l2-ctl --list-devices
720p HD Camera: 720p HD Camera (usb-0000:00:14.0-6):
        /dev/video0
        /dev/video1
        /dev/media0

$ v4l2-ctl -d /dev/video0 --list-ctrls

User Controls

                     brightness 0x00980900 (int)    : min=0 max=127 step=1 default=64 value=64
                       contrast 0x00980901 (int)    : min=0 max=100 step=1 default=50 value=50
                     saturation 0x00980902 (int)    : min=0 max=8 step=1 default=4 value=4
                            hue 0x00980903 (int)    : min=-180 max=180 step=1 default=0 value=0
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=100 max=500 step=1 default=300 value=300
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2 (60 Hz)
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=10 default=3400 value=3400 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=8 step=1 default=4 value=4
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=0 value=0

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=3 (Aperture Priority Mode)
         exposure_time_absolute 0x009a0902 (int)    : min=50 max=10000 step=1 default=166 value=166 flags=inactive
$ v4l2-ctl -d /dev/video1 --list-ctrls
$ v4l2-ctl -d /dev/media0 --list-ctrls
Unable to detect what device /dev/media0 is, exiting.
```

So just a wild uneducated guess: `/dev/video1` is the secondary camera, which needs to be activated (or is just not supported) before using. And `/dev/media0` is the brightness sensor?

I'm kind of stuck, so I thought I'm asking for advice here:

The devices I found (which could be related):

**lspci**

<details>

```console
# lspci -v
[...]
00:12.0 Serial controller: Intel Corporation Tiger Lake-LP Integrated Sensor Hub (rev 20) (prog-if 00 [8250])
        DeviceName: Onboard - Other
        Subsystem: Samsung Electronics Co Ltd Device c194
        Flags: bus master, fast devsel, latency 0, IRQ 16, IOMMU group 8
        Memory at 603e170000 (64-bit, non-prefetchable) [size=64K]
        Capabilities: [80] Power Management version 3
        Capabilities: [90] Vendor Specific Information: Len=14 <?>
        Kernel driver in use: intel_ish_ipc
        Kernel modules: intel_ish_ipc
[...]
00:15.0 Serial bus controller: Intel Corporation Tiger Lake-LP Serial IO I2C Controller #0 (rev 20)
        DeviceName: Onboard - Other
[... 653 lines cut ...]
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval              10
Device Status:     0x0001
  Self Powered
```

</details>

flying-x, 2022-05-17:
Did you check the output of `ddcutil detect`? It might be that the OLED screen doesn't have a backlight that you can control.

I use `icc-brightness-gnome-git` AUR package from arch to simulate brightness change in software.

FedeDP, 2023-03-04:
Hi!
I think perhaps Linux does not support neither your brightness sensor nor your keyboard backlight :)
Not much we can do i'd say!

spikespaz, 2023-12-05:
OLED brightness can be controlled by multiplying each pixel by a brightness factor. This can be done entirely in userspace, for example by GammaStep.

I found another project, `wl-gammarelay-rs` which should support this use-case, but I currently have an open issue because it doesn't work on my machine.

I think Clight should implement something similar, I would come back because now I also have a device with an ambient light sensor.

FedeDP, 2023-12-05:
> I think Clight should implement something similar, I would come back because now I also have a device with an ambient light sensor.

Latest [Clightd tag](https://github.com/FedeDP/Clightd/releases/tag/5.8) supports so-called "software emulated backlight" management for all monitors that are not supported by ddc.
Care to try? I am not sure whether it will work (ie: i am not sure whether your display will try to use sysfs method instead. In that case, i might add a new env var to force-enable emulated backlight mode even for internal sysfs displays).
For more info: https://github.com/FedeDP/Clightd/wiki/Backlight

Also, in Clightd master i disabled by default the emulated backlight support since it was causing some troubles for me when i attached the laptop to my TV, but afaik it worked fine with internal laptop display or my normal external monitor while developing and testing the feature.

spikespaz, 2023-12-05:
For me to use this feature, I would need to be sure it wouldn't try to adjust the (emulated) brightness of two other displays, but rather, only the internal one.

Your concern about it trying to use ACPI sysfs is most likely a problem, `test -f /sys/class/backlight/amdgpu_bl0/brightness` returns `0` so that fine does exist. Writing a value to the file has no visible effect.

Perhaps configure Clight to use emulated backlight for specific displays?

FedeDP, 2023-12-05:
> Perhaps configure Clight to use emulated backlight for specific displays?

Yep makes sense! Please can you open a feat req on Clightd? :) 
Thanks!