FedeDP/Clight #248: Clight not updating gui?

cliffhanger54, 2022-04-25:
Hi

I know you're not responsible for Clight-gui but 
I have installed this on two systems now and all options are greyed out.  Clight is running ok on both systems, but I have nothing in the gui. 
Is Clight reponsible in anyway? 

Thanks

Cliff

FedeDP, 2022-04-25:
Hi! 
The only thing coming up in my mind is that perhaps Clight is not creating the user dbus interface?
You can double check with
`busctl --user introspect org.clight.clight /org/clight/clight`

Thanks!

cliffhanger54, 2022-04-25:
Thanks for that I'm getting:

Failed to introspect object /org/clight/clight of service org.clight.clight: Process org.clight.clight exited with status 1

Looks like answer?

Cliff

FedeDP, 2022-04-25:
I think so :)
Can you share your Clight log in verbose mode?

cliffhanger54, 2022-04-25:
Thanks.  It's not outputing a log file.   I've looked in compilation there are no options there.

FedeDP, 2022-04-25:
You can find the verbose option in Clight conf file (by default /etc/default/clight.conf).
Clight log is instead in $HOME/.local/share/clight/clight.log :)

cliffhanger54, 2022-04-25:
[clight.log](https://github.com/FedeDP/Clight/files/8557147/clight.log)

yeah got it

Cliff

cliffhanger54, 2022-04-27:
So I can take it that the config file shows nothing unusual

My service file in docx form: 
[org.clight.clight.docx](https://github.com/FedeDP/Clight/files/8573370/org.clight.clight.docx)

FedeDP, 2022-06-30:
Sorry for long time without answer!
> [clight.log](https://github.com/FedeDP/Clight/files/8557147/clight.log)
>
> yeah got it
>
> Cliff

This shows interface is actually up!

I see that you are starting Clight from an user systemd service, right? Perhaps Clight is starting before your dbus user session is up?
To quickly test that the issue comes from that, you can try to stop the Clight unit and manually start it.
If then clight-gui gets updated, for me the issue lies in the systemd service :)

cliffhanger54, 2022-06-30:
ThanksI have a feeling this may have something to do with the Wayland windowing system.  Some packages just don't run properly yet.  What do you think?CliffSent from my Huawei Mobile-------- Original Message --------Subject: Re: [FedeDP/Clight] Clight not updating gui? (Issue #248)From: Federico Di Pierro To: FedeDP/Clight CC: cliffhanger54 ,Author 
Sorry for long time without answer!

clight.log
yeah got it
Cliff

This shows interface is actually up!
I see that you are starting Clight from an user systemd service, right? Perhaps Clight is starting before your dbus user session is up?
To quickly test that the issue comes from that, you can try to stop the Clight unit and manually start it.
If then clight-gui gets updated, for me the issue lies in the systemd service :)

[
{
"target": "https://github.com/FedeDP/Clight/issues/248#issuecomment-1171491207",
"url": "https://github.com/FedeDP/Clight/issues/248#issuecomment-1171491207",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

FedeDP, 2022-08-20:
Mmh @cliffhanger54 i do not understand your latest message; perhaps something went wrong while sending it?

FedeDP, 2023-03-04:
Hi! Any news on this topic? Were you able to solve this? :)

cliffhanger54, 2023-03-04:
Sorry the gui never worked. I can only assume that Wayland is the problem.   CliffSent from my Huawei Mobile-------- Original Message --------Subject: Re: [FedeDP/Clight] Clight not updating gui? (Issue #248)From: Federico Di Pierro To: FedeDP/Clight CC: cliffhanger54 ,Mention 
Hi! Any news on this topic? Were you able to solve this? :)

[
{
"target": "https://github.com/FedeDP/Clight/issues/248#issuecomment-1454776462",
"url": "https://github.com/FedeDP/Clight/issues/248#issuecomment-1454776462",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]