FedeDP/Clight #249: [BUG] Clight randomly crashes with  crashes with SIGSEGV on laptop lid close

stoychod, 2022-05-08:
Using Clight version 4.8 on Arch Linux.

The Issue: Clight randomly (not every time) crashes with a  SIGSEGV signal on laptop lid close. The laptop is a newer Lenovo model with an AMD chip that only supports the dreaded _**modern standby**_ and goes  to **_S2idle_** sleep instead of _**S3**_. I don't use a desktop environment or a display manager but a window manager and start Clight in my **_.xinitrc_** (don't know if any of that matters).

Here is my log file:
[clight.log](https://github.com/FedeDP/Clight/files/8647012/clight.log)

And here is the backtrace from GDB:
(gdb) bt
#0  0x00007ffff7b84f7b in ?? () from /usr/lib/libmodule.so.5
#1  0x00007ffff7b85275 in modules_ctx_loop_events () from /usr/lib/libmodule.so.5
#2  0x0000555555560014 in main (argc=1, argv=0x7fffffffe6e8) at /home/stoycho/Build/Clight/src/main.c:46

j4wshua, 2022-06-25:
Can confirm. My laptop is a lenovo model too, but with an intel i5-6200u

FedeDP, 2022-06-25:
Uh sorry! I forgot to answer this!
Thank you very much for the gdb output! I will give a look at it asap!

FedeDP, 2022-06-26:
From the log file and backtrace, i am not able to tell where is the crash :/  
Unfortunately, i never experienced this myself, neither it is reproducible...
Does it ever happened on older versions of Clight? 
@j4wshua are you able to share a log file too? :)
Thank you both! I need more data to track this down basically!

j4wshua, 2022-06-26:
@FedeDP I'll send a log as soon as it happens again!

FedeDP, 2022-06-26:
Yay! Thanks!

stoychod, 2022-06-28:
@FedeDP I am a new user, haven't used and older version. I will give it a try and let you know...

stoychod, 2022-07-08:
@FedeDP  I think I've managed to narrow down the problem. First of all I tried Clight versions 4.7 and .4.3 and the crash still happens with the same backtrace. While testing, though, I observed a few facts:
1. The crash actually happens **not** when closing the lid but when it is **opened** and Clight tries to resume!
2. The crash is **not** random, it happens **only** when I charge the laptop! My workflow is usually as follows: I work on it, the battery runs low, I connect the charger while working then at some point I close the lid and leave the laptop on charge. I  come back after an hour or so, the battery is full so I **first** unplug the charger and **then** open the lid! This the moment the crash happens!
 
 At this point my thinking is that there is some kind of race condition where Clight is too quick to resume while the laptop is in "**undetermined**"  state. So I decided to disable `pause_on_lid_closed` and `capture_on_lid_opened` which I had previously enabled. Now Clight still stops working when I open the lid  but without SIGSEGV fault! Instead it says:
```
(I)[18:09:20]{upower.c:66}	Laptop lid opened.
(I)[18:09:20]{upower.c:55}	AC cable disconnected.
(D)[18:09:20]{interface.c:230}	Emitting 'LidState' property
(D)[18:09:20]{validations.c:175}     Failed to validate lid request.
(D)[18:09:20]{timer.c:32}	Set timeout of 0s 1ns on fd 42.
(D)[18:09:20]{interface.c:230}	Emitting 'AcState' property
```
If I close the lid and open it again or if I manually adjust the brightness it continues to work without a problem until the next charge!
Next I am going to try enabling `resumedelay` to see if that will completely resolve the problem and will let you know!

FedeDP, 2022-08-20:
Hi @stoychod ! You have done a great investigation of the issue!
Did you have any luck with the `resumedelay` option? :)
Thank you very much!

stoychod, 2022-08-23:
Yes, I set `resumedelay=2` and haven't had a crash since! It looks to me that this is a manufacturer/firmware/s2idle implementation problem rather than Clight. I managed to get S3 sleep working, probably I should remove the `resumedelay` option and see if that makes any difference...

FedeDP, 2022-08-28:
That's good to hear! 

> probably I should remove the resumedelay option and see if that makes any difference...
Yeah, it would be great, if you'd like to tinker with it once more :D 
Thank you!

FedeDP, 2022-09-24:
Hi @stoychod ! Any news on this? :)

FedeDP, 2023-03-04:
Closing from now, the issue seemed fixed ;)

closed 2023-03-04 by FedeDP