FedeDP/Clight #250: [question] is it possible that some webcam auto-brighten the images they capture?

flying-x, 2022-05-17:
I am using logitech c922 pro stream webcam. It seems that clight's sensor output is not changing much (between 89%-91%) when I turn on/off lights & blinds in my room. Is it possible that the cam is auto-brightening the images so that clight can't properly work?

Any suggestions on how to find out the potentially issues?

FedeDP, 2022-05-20:
Hi! Yeah that could happen!
Basically, Clightd sets Webcam defaults values before doing any capture. Therefore, if your webcam defaults have auto brightness or some other wizardry active, you'll get the behavior.
But, good thing is Clightd allows you to manually specify configurations for sensors: https://github.com/FedeDP/Clightd/wiki/Sensors#camera-1

Clight passes the Sensor.settings conf option to Clightd for this reason :)

closed 2022-05-20 by flying-x

flying-x, 2022-05-20:
it is working after I manually did the following:
[image]