FedeDP/Clight #251: [BUG] "Timed out waiting on location provided by Geoclue2. Killing module." but geoclue-2.0 is installed?

rlue, 2022-06-26:
**Clight version:**
> 4.7 (built from `master` last night)

**Distribution:**
> Debian bookworm (testing)

**Describe the bug**
Clight can't retrieve location data from Geoclue even though it's installed? Honestly not sure which geoclue package I need here, and if there's any additional setup or configuration required to make it work. I saw [in 2019 you mentioned that there was a Geoclue issues page in the wiki](https://github.com/FedeDP/Clight/issues/73#issuecomment-475148951), but that appears to be gone now?

I'm on Debian and have `geoclue-2.0` and `libgeoclue-2-0` installed.

**Expected behavior**
Clight should succeed in getting location data from geoclue.

**To Reproduce**

* Install from [@brianread108's script](https://github.com/FedeDP/Clight/wiki/Build#dealing-with-dependencies), with prefix `/usr/local`.
* Install Debian apt packages `ddcutil`, `geoclue-2.0`, and `libgeoclue-2-0`.
* Run `clight` as regular user

Not sure if this matters, but I'm on i3wm instead of a more mainstream DE.

```sh
$ clight --verbose
Clightd found, version: 5.4.
AC screen backlight curve: y = -0.024825 + 0.191641x + -0.008928x^2
BL
^
|        **
|      **
|     *
|    *
|   *
|  *
|
| *
|*
|
*---------->BR
BATT screen backlight curve: y = -0.010629 + 0.153844x + -0.007284x^2
BL
^
|
|
|       ***
|     **
|    *
|   *
|  *
| *
|
|*
*---------->BR
Set timeout of 30s 0ns on fd 13.
init_kbd_backlight(): No such file or directory
Keyboard backlight calibration unsupported.
org.freedesktop.ScreenSaver dbus interface exposed.
Disarmed timerfd on fd 35.
AC cable disconnected.
Laptop lid opened.
Emitting 'AcState' property
Disarmed timerfd on fd 38.
Emitting 'LidState' property
Could not find location cache file.
Timed out waiting on location provided by Geoclue2. Killing module.
...
```

nullobsi, 2022-06-30:
https://github.com/FedeDP/Clight/wiki/FAQ#geoclue-asks-for-permission

sometiemes, Geoclue will not let Clight access the location data. You have to give it permission (check link) :)
Hope this helps

rlue, 2022-07-06:
So this did not work, AFAICT. Here are some notes:

* I made sure to `sudo systemctl restart geoclue` and even rebooted, just to be sure.
* When I run `clight` from the terminal, I still get the "Timed out waiting on location provided by Geoclue2. Killing module." error.
* After reading `/etc/geoclue/geoclue.conf` in more detail, I wondered if perhaps I have to launch clight via the .desktop file (`/usr/share/applications/clightc.desktop`) for geoclue to recognize clight / respect the new config setting?

  But I can't figure out how to confirm whether it's working when I launch via the .desktop file—there's no verbose/debug output for me to observe, and under the hood, the .desktop file doesn't even call clight directly:

  ```desktop
  Exec=busctl --expect-reply=false --user call org.clight.clight /org/clight/clight org.clight.clight Capture "bb" false false
  ```

and that's where my knowledge of how this works all falls apart.

In any case, I decided to stop using clight in the end because I just didn't want my webcam indicator light blinking on and off intermittently, so I'm just going to close this issue.

Thanks for a wonderful utility and sorry for the delay on responding to your help. 🙏

closed 2022-07-06 by rlue

rlue, 2022-07-06:
FINAL COMMENT: I didn't have the geoclue agent running. :facepalm: 

Contrary to my assumptions, the geoclue agent is _not_ launched on login via a systemd service; rather, it is launched via `/etc/xdg/autostart/geoclue-demo-agent.desktop`. i3 doesn't handle auto-starting services of any kind; to remedy this, I had to:

```sh
# 1. install dex
$ sudo apt install dex

# 2. symlink geoclue's autostart file to my user autostart directory
$ ln -s /etc/xdg/autostart/geoclue-demo-agent.desktop ~/.config/autostart/

# 3. configure i3 to run dex on login
$ echo "exec --no-startup-id dex --autostart --environment i3 --search-paths $HOME/.config/autostart" >> ~/.config/i3/config
```

I know desktop environment management is not the responsibility of clight documentation per se, but I think it would be nice to have a note about this in the wiki. :)