FedeDP/Clight #252: [FEATURE REQ]

Bonnee, 2022-08-28:
**Is your feature request related to a problem? Please describe.**
My Dell monitor has a firmware bug that kills it upon wakeup, requiring a hard power cycle.

**Describe the solution you'd like**
Inspired by [lunar's FAQ](https://lunar.fyi/faq#firmware-bugs) I propose a similar feature to delay communications after wakeup. I don't know if this will actually fix my issue, but if lunar's got the feature I guess it's working for someone.

**Describe alternatives you've considered**
Since I don't even know if this would work, I haven't thought about alternatives

FedeDP, 2022-08-28:
Ehy! Thanks for opening this issue!
Clight has a [`resumedelay`](https://github.com/FedeDP/Clight/blob/master/Extra/clight.conf#L18) option for issues like this one ;) Care to try?

Bonnee, 2022-08-28:
Ok I totally missed that feature. Seems exactly what I wanted!

closed 2022-08-28 by Bonnee