FedeDP/Clight #253: NixOS daemon not respecting `--conf-file` in user *clight.service*

spikespaz, 2022-09-01:
If you tell me that this isn't a bug with Clight, but rather with the NixOS module, I will republish on the *nixpkgs* repository.

### Issue Description

**TL;DR:** Command with `--conf-file` in *systemd* unit generated by NixOS module starts a service that doesn't respect the provided file, going to default settings. Sometimes.

When using the module provided by *nixpkgs*, enabling the service (plus the *geoclue2* location provider) and using the settings attribute provided has no effect (sometimes). The majority of the time Clight appears to use default settings, and changes made to the configuration file (generated through the Nix configuration) have no effect, even when reloaded with `systemctl --user reload clight.service`.

Note that only once was I able to get Clight to respect the generated configuration. I can confirm that the *systemd* units appear to have been correctly generated and enabled; documentation of this will follow. Additionally, the single instance where Clight behaved as expected was not after the initial install, it happened randomly while I was troubleshooting. I do not know what I changed, but it stopped reading my settings again following a system reboot.

Unfortunately, as it seems impossible to configure Clight at the moment, attaching verbose logs is an impossibility.

#### Environment Specifications
<table>
  <tr>
    <td>Clight version:</td>
    <td>4.8</td>
  </tr>
  <tr>
    <td>Clightd version:</td>
    <td>5.6</td>
  </tr>
  <tr>
    <td>System distribution:</td>
    <td>NixOS 22.11.20220828.a63021a (Raccoon) x86_64</td>
  </tr>
  <tr>
    <td>Configurations:</td>
    <td><a href="https://github.com/spikespaz/dotfiles">https://github.com/spikespaz/dotfiles</a></td>
  </tr>
  <tr>
    <td>Kernel version:</td>
    <td>5.19.4</td>
  </tr>
  <tr>
    <td>Desktop environment:</td>
    <td>Hyprland (Wayland)</td>
  </tr>
  <tr>
    <td>Hardware specifications:</td>
    <td>LENOVO 20Y1S00C00 (ThinkPad P14s Gen 1)</td>
  </tr>
</table>

### Reproduction

Unknown.

### Steps-Taken

I have tried reloading both of the daemons after every ghange, though I believe that this is done by NixOS automatically.

Following are the commands that I have used many times, in many orders, in effort to troubleshoot this myself.

```
$ sudo nixos-rebuild switch --flake path:/home/jacob/dotfiles#
$ sudo systemctl restart clightd.service
$ systemctl --user restart clight.service
$ systemctl --user remove clight.service
$ sudo nixos-rebuild switch --flake path:/home/jacob/dotfiles#
$ nix build --no-link path:/home/jacob/dotfiles#homeConfigurations.jacob.activationPackage
$ "$(nix path-info path:/home/jacob/dotfiles#homeConfigurations.jacob.activationPackage)/activate"
```

### NixOS Module Configuration

**NixOS module source:** <https://github.com/NixOS/nixpkgs/blob/nixos-unstable/nixos/modules/services/x11/clight.nix>
**Unredacted commitish:** <https://github.com/spikespaz/dotfiles/commit/a72d561174f786106b0e59cf55b89e7198442d2f>

```nix
[... 211 lines cut ...]

Restart=on-failure
RestartSec=5
```

#### `systemctl cat clightd.service`

```ini
# /etc/systemd/system/clightd.service
[Unit]
Description=Bus service to manage various screen related properties (gamma, dpms, backlight)
Requires=polkit.service

[Service]
Environment="LOCALE_ARCHIVE=/nix/store/3m17sg8n0066z2a1wqj17ldxzh2vxgva-glibc-locales-2.35-163/lib/locale/locale-archive"
Environment="PATH=/nix/store/mj6wiv9gbbydh06x2i4fl5a9ksdck7c5-coreutils-9.1/bin:/nix/store/nqwkgzay3m6ynr4qz42ypx576s7ckp24-findutils-4.9.0/bin:/nix/store/zpjn33mx0pbfc71ddfi9sc8pdfqrsh8z-gnugrep-3.7/bin:/nix/store/n62963g6w2fj8hnka006wycqwx9pqw5q-gnused-4.8/bin:/nix/store/a026zm6yhqsil81q6f57jgs65ipjgfgz-systemd-251.4/bin:/nix/store/mj6wiv9gbbydh06x2i4fl5a9ksdck7c5-coreutils-9.1/sbin:/nix/store/nqwkgzay3m6ynr4qz42ypx576s7ckp24-findutils-4.9.0/sbin:/nix/store/zpjn33mx0pbfc71ddfi9sc8pdfqrsh8z-gnugrep-3.7/sbin:/nix/store/n62963g6w2fj8hnka006wycqwx9pqw5q-gnused-4.8/sbin:/nix/store/a026zm6yhqsil81q6f57jgs65ipjgfgz-systemd-251.4/sbin"
Environment="TZDIR=/nix/store/2wzs29fwdr03m8a5z5n3mycyay0r73qn-tzdata-2022b/share/zoneinfo"

BusName=org.clightd.clightd
ExecStart=/nix/store/hszqgi3rgd949cswkr2pp3kimdv7w6w3-clightd-5.6/bin/clightd

Restart=on-failure
RestartSec=5
Type=dbus
```

FedeDP, 2022-09-24:
Hi @spikespaz ! Sorry for the very late response! Thanks for the very detailed bug report!
Unfortunately i am not much into nixOS :/ generated systemd units seem fine, and the generated config file for Clight seems ok too!
From my understandings, it seems like a timing issue; just an idea:
> --conf-file /nix/store/vzlqjibfy3s5vq6m7cydxxvifhqv4f9a-clight.conf

Is the file `/nix/store/vzlqjibfy3s5vq6m7cydxxvifhqv4f9a-clight.conf` generated by something else?
Perhaps the clight service starts before that file is generated/accessible from the user, and Clight gracefully discards it!

spikespaz, 2022-12-30:
Sorry, your reply got lost in "read" notifications for three months.

> Is the file `/nix/store/vzlqjibfy3s5vq6m7cydxxvifhqv4f9a-clight.conf` generated by something else?
> Perhaps the clight service starts before that file is generated/accessible from the user, and Clight gracefully discards it!

Nix generates all of these files at generation build time, which I initiate. Programs after subsequent reboots can see all of the depended-on files just as a they would in an orthodox distribution.

FedeDP, 2023-01-04:
> Sorry, your reply got lost in "read" notifications for three months.

No problem :D It happens to me all the time eheh
> Nix generates all of these files at generation build time, which I initiate. Programs after subsequent reboots can see all of the depended-on files just as a they would in an orthodox distribution.

Oh, that is weird; can you try to locally build Clight and forcing verbose=true by default? (eg: by adding a line here: `https://github.com/FedeDP/Clight/blob/master/src/conf/opts.c#L137`, like `conf.verbose=true;`) then you will be able to share a verbose log ;)

infinisil, 2023-04-05:
I'm also using clight on NixOS, but I don't have any problems with the config file getting applied. The only trouble I've had was knowing what changes to make to get the intended result. `verbose = true` works for me too.

FedeDP, 2023-04-06:
Thank you for your input! I never used NixOS therefore I don't really know what should be done and why.
If there's something  to improve the wiki pages please let me know!