FedeDP/Clight #254: Clight not starting

gltchr, 2022-09-12:
Dear Federico, thank you so much for your work with this application!

I've encountered a problem setting it up on the freshly installed Void linux on the laptop, and sadly could not figure it out on my own:

- Clightd service is enabled, running and works (busctl call to +/- brightness results in change in brightness)
- Attempt to to start clight itself (not the demon) fails with _Bad file descriptor... ...is another clight instance running?_
- **htop** search only finds the demon running

**clight -v 4.8**
Let me know if you'll need any particular details about the system...

Any ideas about what I could be doing wrong?
It is a base install without the DE so might be missing some package?
Thanks in advance!

gltchr, 2022-09-14:
ANother re-installation and reboot sequence somehow solved the problem. 
Closing the issue

Once again, thanks for an amazing piece of software!

closed 2022-09-14 by gltchr

FedeDP, 2022-09-14:
Thanks for your kind words :) 
Glad you like Clight!