FedeDP/Clight #255: qustion, dimming ?

promeneur, 2022-09-19:
I get a PC desktop not a laptop.

I try to well understand the clight dimming feature according to be compliant with KDE  "lock screen delay" and "screen switch off delay" features.

"lock screen delay" = 5 mn
"screen switch off delay" = 10 mn

"switching off the screen" <=> to switch the screen from "awake" state to "wake" state. OK ?

On the side of Clight (I use clight-gui) :
In the tab "Dimming"
"AC time out " = 600 s
means after 600 s the screen is dimmed until to be black. Does clight switch off the screen ?

"DPMS AC time out " = 900 s
what does that mean ?

Thanks

FedeDP, 2022-09-24:
Hi! 
`Dimming` refers to "lowering screen backlight when unused", like after 45s that you don't press any key, the screen bl is lowered.
`DPMS` is instead used to put your monitor into standby mode; see: https://wiki.archlinux.org/title/Display_Power_Management_Signaling :)

FedeDP, 2023-03-04:
I think we can close this one; feel free to reopen it if you have any more doubts!

closed 2023-03-04 by FedeDP