FedeDP/Clight #256: [BUG] a black screen instead of the lock screen, a long delay to display the desktop

promeneur, 2022-09-23:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> ...
clight 4.8-b7e0a9b
clightd 5.7-95c7188

**Used distribution:**
> ... openSUSE Tumbleweed

**Describe the bug**
A clear and concise description of what the bug is.

A black screen instead of the lock screen, a long delay to display the desktop

**Expected behavior**
A clear and concise description of what you expected to happen.

A lock screen and no delay to display the desktop

**To Reproduce**
Steps to reproduce the behavior.

First experiment : clight is disabled
1. wait for 5 mn without no action
2. the lock screen is displayed
3. immediately move the mouse cursor
with no delay, the desktop is displayed

Second experiment ; clight is enabled
1. wait for 5 mn without no action
2. randomly, a black screen or the lock screen is displayed
3. immediately move the mouse cursor
you must wait for 15 s to get the desktop

Please attach a Clight verbose log too!

[clight.log](https://github.com/FedeDP/Clight/files/9632686/clight.log)

promeneur, 2022-09-23:
I use X11

FedeDP, 2022-09-23:
Hi! Thanks for opening this issue! Perhaps is the DPMS module kicking in before/after the screenlocker?
Can you try to disable it?

promeneur, 2022-09-23:
now

dpms : 
{
  disabled = true;
  timeouts = [ 900, 300 ];
};

tomorrow i will do some experiments

promeneur, 2022-09-24:
Problem still there.

FedeDP, 2022-09-24:
Weird! Will give this a look and let you know!

FedeDP, 2022-09-24:
Can't see anything wrong in your log file :/ at which time did the monitor switch off happen?

promeneur, 2022-09-25:
A new log
[clight.log](https://github.com/FedeDP/Clight/files/9640461/clight.log)

the phenomenon occurs around 11:51

promeneur, 2022-09-28:
>> Can't see anything wrong in your log file :/ at which time did the monitor switch off happen?

It's not a switch off, just a black screen.

The delay for locking the screen is 5 s

At the moment of the 5 mn is ended there is, very briefly, a black screen then the lock screen.

If you are very speed then you click on the black screen then you get the problem.
If you are less speed then you click on the lock screen then you get immediately the desktop.
If you are slow then you must supply the password, then you get immediately the desktop.

promeneur, 2022-09-28:
The clight.conf

[clight.conf.txt](https://github.com/FedeDP/Clight/files/9663999/clight.conf.txt)

promeneur, 2022-11-07:
Hello

Good news.

The problem disappeared !

Hypothesis :
an update of OS/KDE produced the problem.
an update of OS/KDE solved the problem.

closed 2022-11-07 by promeneur