FedeDP/Clight #257: wiki,opensuse, no more package

promeneur, 2022-09-24:
The package is no more updated.
Last version is : clight 4.5 and clightd 5.3

Then the section "Installation - OpenSuse" is obsolete.

I think the simpliest way to get a clight running is to use the script at the section "Dealing with dependencies" after reading the whole section "Build"

FedeDP, 2022-09-24:
Hi!
I think that having the distro providing Clight is a huge step forward for the end-users! Even if they're a bit outdated :) I can add a warning about it though, thanks!

FedeDP, 2022-11-22:
A note was added in the wiki pages!

closed 2022-11-22 by FedeDP

Kieltux, 2022-11-26:
1. openSUSE (https://build.opensuse.org/package/show/hardware/Clight) does offer Clight 4.8. And Clight 4.9 is on the way.
2. openSUSE (https://build.opensuse.org/package/show/hardware/Clightd) does offer Clightd 5.6. And Clightd 5.7 is on the way.
Both packages are available for openSUSE Tumbleweed, openSUSE Leap 15.3, openSUSE Leap 15.4 and more.

FedeDP, 2022-11-26:
Wow this is huge. Thanks for the update! I am going to revert the note on the Wiki! :rocket: