FedeDP/Clight #258: [FEATURE REQ] being consistent with KDE settings

promeneur, 2022-09-25:
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

KDE manages : 
- eye damage by blue light emitted by LEDs with "Night color"
- lock screen with 5 mn delay by default
- screen saver by switching off (switch to standby) the monitor after a 10 mn delay by default

All these features are enabled by default, excepted "Night Color" which requires a user action.
To be consistent with KDE, when installing Clight or running Clight, it is necessary to disable KDE features if Clight gets the same one.
Also it is consistent for Clight : 
- to dim just before the lock screen
- to switch off the screen after the lock screen

Why ?
- because it is not good that there are two services which manage the same thing. This leads to problems.
- it is a good thing there is only one spot for monitor settings

The consistency must be handled by Clight not the user because the user is mainly not a techie person.

Thanks for your attention.

FedeDP, 2022-12-28:
Hi!
I actually think it is a bit out of scope for Clight (and a bit too hard given its little manpower, ie: me with less and less spare time :smile:  )
I think a small wiki page would be enough; would you mind working on that? You already have repo access eheh

promeneur, 2022-12-31:
I suggest an easy way with KDE.

By default, don't switch off the screen, let KDE manage the feature “screen energy saving “switch off the screen after a delay””

That's all. KDE is a poor tool to manage the screen.

For “suspend to ram or disk” I assume this is the OS which manages these features and switch off the screen.

FedeDP, 2023-01-02:
> By default, don't switch off the screen, let KDE manage the feature “screen energy saving “switch off the screen after a delay””

Yep but then all the people using Clight in a simpler environment (like no DE at all, just wm), would have to tinker with Clight conf file; i don't like that from an UX perspective.
Perhaps i can try to manually set default value for DPMS, DIMMER to on or off by checking at runtime if Clight is running in a DE; wdyt?

> For “suspend to ram or disk” I assume this is the OS which manages these features and switch off the screen.

Note that Clight does not manage Suspend, it just manages DPMS (that is different, it just switches off the screen but does not suspend the system).

FedeDP, 2023-01-02:
Note however that i am not that convinced about this feature; it is disruptive for people that today run Clight with default settings on DEs.
I still think that best is for Clight to default to enable everything, and just let user disable features she/he does not need.

FedeDP, 2023-03-04:
I am going to close this one; i think it is out of scope for Clight (i am spring-cleaning issues :D ) 
Feel free to reopen if you still got any doubt! Thank you!

closed 2023-03-04 by FedeDP