FedeDP/Clight #259: [question]Pause/resume brightness adjusting, Set gamma to max/min gamma in a script ?

promeneur, 2022-10-04:
The script I wrote using the statements you supplied to me in the past
********************************************************************************************************************************************
#!/bin/bash
#
#Pause brightness adjusting
busctl --expect-reply=false --user call org.clight.clight /org/clight/clight org.clight.clight Pause "b" true

#Set gamma to max gamma
busctl --user set-property org.clight.clight /org/clight/clight/Conf/Gamma org.clight.clight.Conf.Gamma NightTemp "i" 6500

/usr/bin/cvlc

#Restore gamma to min gamma 
busctl --user set-property org.clight.clight /org/clight/clight/Conf/Gamma org.clight.clight.Conf.Gamma NightTemp "i" 3500

#Resume brightness adjusting
busctl --expect-reply=false --user call org.clight.clight /org/clight/clight org.clight.clight Pause "b" false
**************************************************************************************************************************************************
It seems the statement has no more effect.

What are the good ones ?

Thanks

FedeDP, 2022-11-04:
Hi!
Sorry for the late response!
That is weird; it should work! Can you share the clight.log while running the script?
Thank you!

promeneur, 2022-11-22:
Here is the log when launching then quitting VLC
*****************************************************************************
D)[14:46:47]{timer.c:32}	Set timeout of 30s 0ns on fd 42.
(D)[14:46:47]{interface.c:230}	Emitting 'ScreenComp' property
(D)[14:47:14]{pm.c:204}	Suspend ctr: 1
(D)[14:47:14]{utils.c:71}	Pausing BACKLIGHT: SUSPEND
(D)[14:47:14]{utils.c:71}	Pausing DPMS: SUSPEND
(D)[14:47:14]{utils.c:71}	Pausing SCREEN: SUSPEND
(D)[14:47:14]{utils.c:71}	Pausing GAMMA: SUSPEND
(D)[14:47:14]{interface.c:230}	Emitting 'Suspended' property
(D)[14:47:14]{utils.c:71}	Pausing DIMMER: SUSPEND
(D)[14:48:01]{pm.c:204}	Suspend ctr: 0
(D)[14:48:01]{utils.c:77}	Resuming BACKLIGHT: SUSPEND
(D)[14:48:01]{utils.c:77}	Resuming DPMS: SUSPEND
(D)[14:48:01]{utils.c:77}	Resuming SCREEN: SUSPEND
(D)[14:48:01]{utils.c:77}	Resuming GAMMA: SUSPEND
(D)[14:48:01]{interface.c:230}	Emitting 'Suspended' property
(D)[14:48:01]{utils.c:77}	Resuming DIMMER: SUSPEND
(D)[14:48:01]{screen.c:149}	Average screen-emitted brightness: 0.010314.
(D)[14:48:01]{timer.c:32}	Set timeout of 30s 0ns on fd 42.
(D)[14:48:01]{interface.c:230}	Emitting 'ScreenComp' property
**********************************************************************************************

It seems that messages changed. In the past, there were explicit messages as:
- gamma set to 6500
- gamma set to 3500
etc.

I want the true colors (colors without any filter) when using VLC or Digikam.
When brightness management is suspended, is brightness really set to full brightness ?
When gamma management is suspended, is gamma really set to 6500 ?

FedeDP, 2022-11-22:
I think this might be a new script:
```
#Pause brightness adjusting
busctl --expect-reply=false --user set-property org.clight.clight /org/clight/clight/Conf/Backlight org.clight.clight.Conf.Backlight NoAutoCalib "b" true

# Set max backlight level
busctl --user --expect-reply=false call org.clight.clight /org/clight/clight org.clight.clight IncBl "d" 1.0

#Set gamma to max gamma
busctl --user set-property org.clight.clight /org/clight/clight/Conf/Gamma org.clight.clight.Conf.Gamma NightTemp "i" 6500

/usr/bin/cvlc

#Restore gamma to min gamma
busctl --user set-property org.clight.clight /org/clight/clight/Conf/Gamma org.clight.clight.Conf.Gamma NightTemp "i" 3500

# Force a backlight calibration
busctl --expect-reply=false --user call org.clight.clight /org/clight/clight org.clight.clight Capture "bb" false false

#Resume brightness adjusting
busctl --expect-reply=false --user set-property org.clight.clight /org/clight/clight/Conf/Backlight org.clight.clight.Conf.Backlight NoAutoCalib "b" false
```

I cannot recall why i shared the script you shared :/

promeneur, 2022-11-23:
I tested with digikam the brightness adjusting (gamma adjusting is currently disabled). Here is the result when launching (10;27)and quitting (10:29) digikam.
*************************************************************************************
(I)[10:15:48]{backlight.c:455}	Ambient brightness: 0.809 (-0.017 screen compensation) -> Screen backlight pct: 0.933.
(I)[10:25:49]{backlight.c:455}	Ambient brightness: 0.826 (-0.018 screen compensation) -> Screen backlight pct: 0.941.
(I)[10:27:22]{backlight.c:560}	Backlight autocalibration disabled.
(I)[10:29:33]{backlight.c:455}	Ambient brightness: 0.836 (-0.022 screen compensation) -> Screen backlight pct: 0.944.
(I)[10:29:33]{backlight.c:560}	Backlight autocalibration enabled.
*************************************************************************************

There is no message saying "brightness is set to max. Is it normal ?

promeneur, 2022-11-23:
I made this test :
- I launch digikam
- i launc a video call with skype
- during videocall i quit digikam
ambient brightness adjusting is not restarting immediately

******************************************************************************************
Screen backlight pct: 0.944.
(I)[10:29:33]{backlight.c:560}	Backlight autocalibration enabled.
(I)[10:35:50]{backlight.c:455}	Ambient brightness: 0.827 (-0.015 screen compensation) -> Screen backlight pct: 0.943.
(I)[10:37:54]{backlight.c:560}	Backlight autocalibration disabled.
(I)[11:00:21]{backlight.c:560}	Backlight autocalibration enabled.
******************************************************************************************

FedeDP, 2023-03-04:
> There is no message saying "brightness is set to max. Is it normal ?

Is brightness set to max indeed?
Is this working?

>  ambient brightness adjusting is not restarting immediately

This is ok, because during a video call brightness adjusting is not working since camera is busy.

promeneur, 2023-03-05:
I checked there si no more problem.

closed 2023-03-05 by promeneur