FedeDP/Clight #260: [BUG]

GiGGioSo, 2022-11-04:
Hi! I'm using `clight 4.8-2` and `clightd 5.6-1` on ArchLinux (`linux 6.0.6.arch1-1`)

**Describe the bug**
Starting `clight` the camera doesn't get activated, and in the clight output there is `capture_frames_brightness(): Input/output error`.

I tried looking around in the wiki, and tried a couple of busctl commands:
- `busctl call org.clightd.clightd /org/clightd/clightd/Sensor/Camera org.clightd.clightd.Sensor IsAvailable "s" ""`
returns
`sb "/dev/video0" true`

**but**
- `busctl call org.clightd.clightd /org/clightd/clightd/Sensor/Camera org.clightd.clightd.Sensor Capture "sis" "" 5 ""`
returns
`Call failed: Input/output error`

**P.S.**: I recently switched from `pulseaudio` to `pipewire` for the bluetooth, could this have something to do with it?

**Verbose clight log**
```
Clight
* Software version:		4.8-fc3df9c
* Global config dir:		/etc/default
* Global data dir:		/usr/share/clight
* Starting time:		Fri Nov  4 10:16:31 2022

Starting options:

### GENERIC ###
* Verbose (debug):		Enabled
* ResumeDelay:		0

### BACKLIGHT ###
* Smooth trans:		Enabled
* Smooth step:		0.05
* Smooth timeout:		30
* Smooth fixed:		0
* Daily timeouts:		AC 30	BATT 60
* Nightly timeouts:		AC 30	BATT 60
* Event timeouts:		AC 30	BATT 60
* Shutter threshold:		0.00
* Autocalibration:		Enabled
* Pause on lid closed:		Disabled
* Capture on lid opened:		Disabled
* Restore On Exit:		Disabled

### SENSOR ###
* Captures:		AC 5	BATT 5
* Device:		Unset
* Settings:		Unset

### KEYBOARD ###
* Dim:		Disabled
* Timeouts:		AC 15	BATT 5

### GAMMA ###
* Smooth trans:		Enabled
* Smooth steps:		50
* Smooth timeout:		300
* Daily screen temp:		6500
* Nightly screen temp:		4000
* Long transition:		Disabled
* Ambient gamma:		Disabled
* Restore On Exit:		Disabled

### DAYTIME ###
* User position:		Unset
* User set sunrise:		Unset
* User set sunset:		Unset
* Event duration:		1800
* Sunrise offset:		0
* Sunset offset:		0

### DIMMER ###
* ENTER Smooth trans:		Enabled
* ENTER Smooth step:		0.05
* ENTER Smooth timeout:		30
* ENTER Smooth fixed:		0
* EXIT Smooth trans:		Enabled
* EXIT Smooth step:		0.05
* EXIT Smooth timeout:		30
* EXIT Smooth fixed:		0
* Timeouts:		AC 0	BATT 60
* Backlight pct:		0.15

### DPMS ###
* Timeouts:		AC 900	BATT 300

### SCREEN ###
* Timeouts:		AC 30	BATT -1
* Contrib:		0.10
* Samples:		10

### INHIBIT ###
* Docked:		Disabled
* PowerManagement:		Disabled
* Backlight:		Disabled

(I)[10:16:31]{main.c:145}	Clightd found, version: 5.6-77b1953.
(D)[10:16:31]{my_math.c:73}	AC screen backlight curve: y = 0.047413 + 0.131816x + -0.005536x^2
BL
^
|          
|          
|        **
|      **  
|    **    
|   *      
|  *       
| *        
|*         
*          
+---------->BR
(D)[10:16:31]{my_math.c:73}	BATT screen backlight curve: y = 0.060699 + 0.034443x + 0.000956x^2
BL
^
|          
|          
|          
|          
|          
|         *
|       ** 
|    ***   
|  **      
***        
+---------->BR
(D)[10:16:31]{timer.c:32}	Set timeout of 30s 0ns on fd 15.
(D)[10:16:31]{bus.c:189}	init_kbd_backlight(): No such file or directory
(I)[10:16:31]{keyboard.c:158}	Keyboard backlight calibration unsupported.
(I)[10:16:31]{interface.c:181}	org.freedesktop.ScreenSaver dbus interface exposed.
(D)[10:16:31]{timer.c:34}	Disarmed timerfd on fd 37.
(I)[10:16:31]{upower.c:55}	AC cable disconnected.
(I)[10:16:31]{upower.c:66}	Laptop lid opened.
(D)[10:16:31]{interface.c:230}	Emitting 'AcState' property
(D)[10:16:31]{timer.c:34}	Disarmed timerfd on fd 40.
(D)[10:16:31]{interface.c:230}	Emitting 'LidState' property
(D)[10:16:31]{location.c:152}	Could not find location cache file.
(D)[10:16:31]{bus.c:189}	geoclue_ping(): The name org.freedesktop.GeoClue2 was not provided by any .service files
(D)[10:16:31]{location.c:174}	Geoclue2 is not present.
(D)[10:16:31]{timer.c:32}	Set timeout of 0s 1ns on fd 15.
(W)[10:16:31]{location.c:86}	Failed to init (no geoclue2 present?). Killing module.
(W)[10:16:31]{daytime.c:72}	No location provider found; fallback to DAY daytime.
(D)[10:16:31]{interface.c:230}	Emitting 'Location' property
(W)[10:16:31]{gamma.c:91}	Killing GAMMA as no location provider is available.
(D)[10:16:31]{interface.c:230}	Emitting 'DayTime' property
(D)[10:16:31]{timer.c:32}	Set timeout of 0s 1ns on fd 13.
(D)[10:16:31]{backlight.c:403}	Sensor '/dev/video0' is now available.
(D)[10:16:31]{backlight.c:663}	Resumed as a sensor is now available.
(D)[10:16:31]{interface.c:230}	Emitting 'SensorAvail' property
(D)[10:16:31]{bus.c:189}	capture_frames_brightness(): Input/output error
(D)[10:16:31]{timer.c:32}	Set timeout of 60s 0ns on fd 13.
(I)[10:16:38]{signal.c:48}	Received 2. Leaving.
```

FedeDP, 2022-11-04:
Hi! Thanks for opening this bug report.
I am indeed seeing the same issue myself; i think it has something to do with latest kernel release.
I'd wait a couple of days for an upstream fix; otherwise i'll look into it :)

FedeDP, 2022-11-04:
> P.S.: I recently switched from pulseaudio to pipewire for the bluetooth, could this have something to do with it?

I don't think so; Clightd manually opens /dev/videoX devices and manages them through v4l2 kernel API.

GiGGioSo, 2022-11-05:
Just to put it out there, still not working with `linux 6.0.7.arch1-1`.

FedeDP, 2022-11-08:
Hi! Latest commit from Clightd master fixed the issue for me; care to try? Just install clightd-git package :)

FedeDP, 2022-11-08:
It seems like some other daemon (i think pipewire) is now setting an INTERACTIVE priority on all cameras; Clight used to set `background` prio, ie: lowest possible level; but if the camera is opened with an higher priority by any other application, you are not able to modify nothing from a lower priority application.
See: https://docs.kernel.org/userspace-api/media/v4l/vidioc-g-priority.html#vidioc-g-priority for reference!

GiGGioSo, 2022-11-08:
Hey! I installed `clightd-git` but nothing seems to have changed :(.

**Regarding PipeWire**
I'll start saying I don't know much, in fact I installed PipeWire a week ago just to be able to use the aptX codec for my bluetooth headset.

This being said, nothing should be using the camera *(also because there's a led that lights up whenever the cam is used, and it just doesn't)*.

Also because, looking at the [archwiki](https://wiki.archlinux.org/title/PipeWire#Video) it looks like I would need the `pipewire-v4l2` packet to be able to manage the camera and stuff, and I don't have that packet.

Thanks for the help and for having built clight(d)!

FedeDP, 2022-11-08:
Did you rebooted your system after upgrading clightd?

GiGGioSo, 2022-11-08:
Full my bad, I assumed restarting Xorg would be enough.

Now it works! :)

Thank you so much for everything again!!!!

P.S.: I'm the one supposed to close the issue?

FedeDP, 2022-11-08:
Yep i think we can close this one! I'll proceed to release a new clightd version later this week hopefully!

closed 2022-11-08 by GiGGioSo