FedeDP/Clight #262: [BUG] trans_timeout lower that 0.001 break backlight dimming

jclsn, 2022-11-17:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> 4.9

**Used distribution:**
> Manjaro

**Describe the bug**
When I set a value lower than 0.001, clight is dimming independently without actually dimming.

**Expected behavior**
It should dim

**To Reproduce**
Set a ``trans_timout`` lower than 0.001 and enable smooth dimming.

```
Clightd found, version: 5.7-dd12e8d.
AC screen backlight curve: y = 0.005804 + 0.009009x + 0.008508x^2
BL
^
|          
|         *
|        * 
|          
|       *  
|      *   
|     *    
|    *     
|   *      
| **       
**--------->BR
BATT screen backlight curve: y = -0.010629 + 0.153844x + -0.007284x^2
BL
^
|          
|          
|       ***
|     **   
|    *     
|   *      
|  *       
| *        
|          
|*         
*---------->BR
Set timeout of 30s 0ns on fd 18.
Keyboard backlight calibration supported.
AC keyboard backlight curve: y = 0.574825 + -0.195944x + 0.014685x^2
BL
^
|          
|          
|          
|          
*          
|          
|*         
|          
| *        
|  *      *
+---******->BR
BATT keyboard backlight curve: y = 0.574825 + -0.195944x + 0.014685x^2
BL
^
|          
|          
|          
|          
*          
|          
|*         
|          
| *        
|  *      *
+---******->BR
Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
Monitoring requests to org.freedesktop.ScreenSaver name owner.
Disarmed timerfd on fd 42.
AC cable connected.
Laptop lid opened.
Emitting 'AcState' property
Set timeout of 5s 0ns on fd 43.
Emitting 'LidState' property
Emitting 'ScreenBr' property
53.54 9.99 loaded from cache file.

Loc distance: 53.54,9.99 -> 91.00,181.00 : 2451.05 km.
New location received: 53.54, 9.99.
Set timeout of 0s 1ns on fd 44.
Emitting 'Location' property
Next alarm due to: Fri Nov 18 07:21:00 2022
Set timeout of 51591s 0ns on fd 44.
Emitting 'Sunrise' property
Emitting 'Sunset' property
Emitting 'DayTime' property
Emitting 'NextEvent' property
Set timeout of 0s 1ns on fd 45.
Sensor '/dev/video0' is now available.
Resumed as a sensor is now available.
Emitting 'SensorAvail' property
Captured [5/5] from '/dev/video0'. Ambient brightness: 0.051.
Set timeout of 2700s 0ns on fd 45.
Emitting 'AmbientBr' property
Emitting 'ScreenBr' property
Content calib: wmax: 0.213, wmin: -0.187, new_bl: 0.181
Ambient brightness: 0.051, Screen brightness: 0.078 -> Screen backlight: 0.181.
Screen backlight: 0.181 -> Keyboard backlight: 0.268.
Emitting 'BlPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'KbdPct' property
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
53.56 9.96 received from Geoclue2.
Loc distance: 53.56,9.96 -> 53.54,9.99 : 2.32 km.
Failed to validate location request.
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
Backlight 'intel_backlight' level updated: 0.19.
Emitting 'BlPct' property
Emitting 'KbdPct' property
^CReceived 2. Leaving.
Latest location stored in cache file!
```

FedeDP, 2022-11-22:
Hi! Thanks for reporting this issue!

> Maybe the steps are too small for the hardware?

I guess so; Clightd sets the new step as `current pct + increment_pct * max_bl_level`; if `max_bl_level` is too low, increment_pct * max_bl_level will round to 0 and current pct will never change.

FedeDP, 2022-11-22:
PS: in my opinion, Clight on laptop behaves wonderfully when using the `trans_fixed` dimming option: https://github.com/FedeDP/Clight/blob/master/Extra/clight.conf#L70

jclsn, 2022-11-24:
Yeah, that works quite well indeed. I will just use this now. Thanks

closed 2022-11-24 by jclsn