FedeDP/Clight #263: [FEATURE REQ] Custom stepping for when playing videos (inhibited)

emansom, 2022-11-18:
**Describe the solution you'd like**
When watching videos (inhibited), I'd like Clight to optionally allow a custom stepping to be defined.

This'd allow having only e.g. the highest brightness for videos, and lower brightness for regular desktop usage or vice-versa.

FedeDP, 2022-11-19:
Hi! 
Thank for opening this feature request! The feature itself is not available in Clight, but you can easily achieve it through DBus API or custom modules; here is an example of a custom module: https://github.com/FedeDP/Clight/wiki/Custom-Modules#inhibit-backlight.
Note: the skeleton module is shipped with Clight, instlled in `/usr/share/clight/inhibit_bl.skel`.

FedeDP, 2023-06-07:
Closing this one.
Feel free to reopen if you need further help!