FedeDP/Clight #268: Enable/disable gamma correction at night

mmguar, 2022-12-28:
I might have missed the ability to do so in clight/clightd, but I was not able to find a way to restore gamma to DayTemp at night, which is a common use case, for example when watching a movie.
I used dbus to change NightTemp to the value of DayTemp as a quick solution, and I added the functionality to the gui, but it would be nice to have a boolean NightPause property available through dbus, as other solutions require storing the original value of NightTemp, to be able to restore it.

FedeDP, 2023-03-04:
Hi! Latest master exposes `org.clight.clight.Conf.Gamma.Toggle` method to toggle between states' temps, like:
`busctl --user call org.clight.clight /org/clight/clight/Conf/Gamma org.clight.clight.Conf.Gamma Toggle`
;)

FedeDP, 2023-03-04:
Moreover, desktop file exposes a ToggleGamma action now!

closed 2023-03-04 by FedeDP

FedeDP, 2023-03-04:
Wiki page has been updated accordingly: https://github.com/FedeDP/Clight/wiki/Api#bus-methods-3

commit 387d9a25 (2023-03-04): new(gamma): add an exposed Toggle method to Conf.Gamma dbus API.

It toggles between states' temps. Fixes #268.