FedeDP/Clight #270: error - Could not create dbus interface

taylon, 2023-01-22:
**Clight version the issue has been seen with:**
> 4.9-8069945

**Used distribution:**
> Artix

When I run clight the following errors (full verbose log is attached):

```
Could not create dbus interface 'org.clight.clight.Conf.Backlight': Invalid argument
Could not create dbus interface 'org.clight.clight.Conf.Sensor': Invalid argument
Could not create dbus interface 'org.clight.clight.Conf.MonitorOverride': Invalid argument
Could not create dbus interface 'org.clight.clight.Conf.Inhibit': Invalid argument
Could not create dbus interface 'org.clight.clight.Conf.Dpms': Invalid argument
Could not create dbus interface 'org.clight.clight.Conf.Daytime': Invalid argument
Keyboard backlight calibration supported.
Could not create dbus interface 'org.clight.clight.Conf.Kbd': Invalid argument
BUS: Failed to connect to user bus
Could not create dbus interface 'org.clight.clight.Conf.Screen': Invalid argument
Could not create dbus interface 'org.clight.clight.Conf.Gamma': Invalid argument
Could not create org.clight.clight dbus interface: Channel number out of range
Failed to init. Killing module.
Could not create dbus interface 'org.clight.clight.Conf.Dimmer': Invalid argument
```

clightd is running but aparrently no module is loaded and so nothing happens.

If I run a commang like `busctl --user call org.clight.clight /org/clight/clight org.clight.clight IncBl d 0.1`  I get `Failed to connect to bus: No such file or directory`.

Not sure if it is relevant but I'm running with openrc (so no systemd). I also have a pretty minimal Artix install in general so I might be missing something important installed in the system.

[clight.log](https://github.com/FedeDP/Clight/files/10473600/clight.log)

FedeDP, 2023-01-22:
Hi! Thanks for opening this issue!
It seems like your session is not launching dbus; you shoud be able to run one using: https://linux.die.net/man/1/dbus-launch

taylon, 2023-01-26:
hm... looks like it is running:

`busctl list`
```
NAME                          PID PROCESS     USER    CONNECTION    UNIT    SESSION DESCRIPTION
:1.0                         1417 elogind     root    :1.0          elogind -       -          
:1.1                         1981 iwd         root    :1.1          iwd     -       -          
:1.18                        4219 busctl      taylon  :1.18         1       -       -          
:1.3                         2401 Xorg        taylon  :1.3          1       -       -          
:1.5                         2418 clightd     root    :1.5          dbus    -       -          
:1.6                         2423 upowerd     root    :1.6          dbus    -       -          
:1.7                         2428 polkitd     polkitd :1.7          dbus    -       -          
:1.8                         2558 qutebrowser taylon  :1.8          1       -       -          
net.connman.iwd              1981 iwd         root    :1.1          iwd     -       -          
org.clightd.clightd          2418 clightd     root    :1.5          dbus    -       -          
org.freedesktop.DBus         1390 dbus-daemon root    -             dbus    -       -          
org.freedesktop.PolicyKit1   2428 polkitd     polkitd :1.7          dbus    -       -          
org.freedesktop.RealtimeKit1    - -           -       (activatable) -       -       -          
org.freedesktop.UPower       2423 upowerd     root    :1.6          dbus    -       -          
org.freedesktop.login1       1417 elogind     root    :1.0          elogind -       -          
```

`busctl status`:
```
BusAddress=unix:path=/run/dbus/system_bus_socket
BusScope=system
BusID=6f1713f27d38915cb942464363d1f08f
PID=1390
PPID=1389
TTY=n/a
UID=81
EUID=0
SUID=81
FSUID=81
GID=81
EGID=0
SGID=81
FSGID=81
SupplementaryGIDs=
Comm=dbus-daemon
CommandLine=/usr/bin/dbus-daemon --system --nofork --nopidfile
CGroup=/dbus
Session=dbus
AuditLoginUID=n/a
AuditSessionID=n/a
EffectiveCapabilities=cap_audit_write
PermittedCapabilities=cap_audit_write
InheritableCapabilities=
BoundingCapabilities=cap_chown cap_dac_override cap_dac_read_search 
        cap_fowner cap_fsetid cap_kill cap_setgid 
        cap_setuid cap_setpcap cap_linux_immutable cap_net_bind_service 
        cap_net_broadcast cap_net_admin cap_net_raw cap_ipc_lock 
        cap_ipc_owner cap_sys_module cap_sys_rawio cap_sys_chroot 
        cap_sys_ptrace cap_sys_pacct cap_sys_admin cap_sys_boot 
        cap_sys_nice cap_sys_resource cap_sys_time cap_sys_tty_config 
        cap_mknod cap_lease cap_audit_write cap_audit_control 
        cap_setfcap cap_mac_override cap_mac_admin cap_syslog 
        cap_wake_alarm cap_block_suspend cap_audit_read cap_perfmon 
        cap_bpf cap_checkpoint_restore
```

`rc-status`:
```
Runlevel: default
 dbus                                        [  started 00:40:53 (0) ]
 iwd                                         [  started 00:40:53 (0) ]
 netmount                                                [  stopped  ]
 ntpd                                        [  started 00:40:53 (0) ]
 local                                                   [  started  ]
 agetty.tty4                                 [  started 00:40:53 (0) ]
 agetty.tty3                                 [  started 00:40:53 (0) ]
 agetty.tty6                                 [  started 00:40:53 (0) ]
 agetty.tty2                                 [  started 00:40:53 (0) ]
 agetty.tty5                                 [  started 00:40:53 (0) ]
 agetty.tty1                                 [  started 00:40:53 (0) ]
Dynamic Runlevel: hotplugged
Dynamic Runlevel: needed/wanted
Dynamic Runlevel: manual
```

any other ideas? or am I looking at the wrong thing?

FedeDP, 2023-01-26:
Yep, i am talking about the user dbus session; can you try: `busctl --user status`?

taylon, 2023-01-27:
That was it!
I added `dbus-launch clight &` to my xinitrc and it worked great.
Thank you very much :)

closed 2023-01-27 by taylon