FedeDP/Clight #271: [BUG] Invalid desktop file

ferki, 2023-02-08:
**Clight version the issue has been seen with:**

4.9

**Used distribution:**

Gentoo

**Describe the bug**

Part of the installation process on Gentoo is to run `.desktop` files through QA checks, and it complains about validation errors (as reported via ferki/gentoo-overlay#3; see [relevant build.log entries](https://gist.github.com/asarubbo/afdaaf4f565dff698c3eeb69757b51b0#file-build-log-L657-L662)).

Running the failing desktop file through `desktop-file-validate` explains the error as:

> key "Exec" in group "Desktop Action ForceGamma" contains a reserved character '&' outside of a quote

which is also potentially a bug, since the `Exec` key might not be able to run both specified commands as expected.

See also:

- [The `Exec` key](https://specifications.freedesktop.org/desktop-entry-spec/latest/ar01s07.html) specification
- <details><summary>full validation output</summary>

    ```bash
    $ desktop-file-validate /usr/share/applications/clightc.desktop
    /usr/share/applications/clightc.desktop: error: value "busctl --expect-reply=false --user set-property org.clight.clight /org/clight/clight/Conf/Gamma org.clight.clight.Conf.Gamma DayTemp "i" 0 && busctl --expect-reply=false --user set-property org.clight.clight /org/clight/clight/Conf/Gamma org.clight.clight.Conf.Gamma NightTemp "i" 0" for key "Exec" in group "Desktop Action ForceGamma" contains a reserved character '&' outside of a quote
    /usr/share/applications/clightc.desktop: error: value "busctl --expect-reply=false --user set-property org.clight.clight /org/clight/clight/Conf/Gamma org.clight.clight.Conf.Gamma DayTemp "i" 0 && busctl --expect-reply=false --user set-property org.clight.clight /org/clight/clight/Conf/Gamma org.clight.clight.Conf.Gamma NightTemp "i" 0" for key "Exec" in group "Desktop Action ForceGamma" contains a reserved character '&' outside of a quote
    ```
</details>

**Expected behavior**

Desktop files installed by clight must be valid.

**To Reproduce**

Run `desktop-file-validate Extra/desktop/clightc.desktop` in a cloned git repo.

Not attaching a Clight verbose log, as it seems irrelevant for this bug. Let me know if you still need those details.

FedeDP, 2023-03-02:
Hi! Thanks for opening this issue, and great catch!
Are you willing to open a PR to fix the issue?

ferki, 2023-03-03:
> Are you willing to open a PR to fix the issue?

In general, yes, but I have no idea which way of solving this would be ideal or preferred, so I'd need guidance – especially given that I don't use this part of clight at all, so I'm not sure how to test :)

The few ideas that go through my mind at first glance:

- Fix the quoting somehow?
- Try to escape invalid characters?
- Should there be a combined `SetTemp` instead of having to call both `DayTemp` and `NightTemp`?
- Is there already something we can call instead just like the imaginary `SetTemp` above?

FedeDP, 2023-03-03:
Given the exec spec, i'd say this should work:
```
sh -c "busctl --expect-reply=false --user set-property org.clight.clight /org/clight/clight/Conf/Gamma org.clight.clight.Conf.Gamma DayTemp "i" 0 && busctl --expect-reply=false --user set-property org.clight.clight /org/clight/clight/Conf/Gamma org.clight.clight.Conf.Gamma NightTemp "i" 0"
```

With this patch, `desktop-file-validate` does not complain anymore and the desktop file works fine ;) 
I will push the change!

closed 2023-03-03 by FedeDP

ferki, 2023-03-03:
Oh, yeah, that might work too, thanks! :)

I can confirm that `desktop-file-validate` is not complaining anymore :+1: 

I wonder whether `sh` gets confused by all the double quotes in there or not, though.

FedeDP, 2023-03-03:
I tested and it worked for me :)

commit 995411ca (2023-03-03): fix(extra): fixed clightc desktop file validation.

Fixes #271