FedeDP/Clight #272: [BUG] calibration wizard fails with "get_backlight(): No such device or address"

zootboy, 2023-02-08:
**Clight version the issue has been seen with:**
> 4.9-8069945

**Used distribution:**
> Arch Linux
> Framework Laptop

**Describe the bug**
Running the config wizard fails after the manual backlight adjustment step:

```
$ clight -w
Config file /home/sean/.config/clight.conf not found.
INHIBIT_CONF: not needed. Disabling.
Welcome to Clight wizard. Press ctrl-c to quit at any moment.
Wizard will use 'intel_backlight' screen.
Waiting for sensor...
Set timeout of 0s 1ns on fd 20.
Sensor '/dev/iio:device0' is now available.
Resumed as a sensor is now available.
Pausing BACKLIGHT: AUTOCALIB
Sensor available. Start? [Y/n]: > 

Move to nightlight-like light.
Are you ready? [Y/n]

Captured [5/5] from '/dev/iio:device0'. Ambient brightness: 0.135.
Ambient brightness is currently 0.135; redo capture? [y/N]: > 
Set desired screen backlight then press ENTER... > Backlight 'intel_backlight' level updated: 1.00.
Backlight 'intel_backlight' level updated: 0.95.
Backlight 'intel_backlight' level updated: 0.90.
Backlight 'intel_backlight' level updated: 0.85.
Backlight 'intel_backlight' level updated: 0.80.
Backlight 'intel_backlight' level updated: 0.75.
Backlight 'intel_backlight' level updated: 0.70.
Backlight 'intel_backlight' level updated: 0.65.
Backlight 'intel_backlight' level updated: 0.60.
Backlight 'intel_backlight' level updated: 0.55.
Backlight 'intel_backlight' level updated: 0.50.
Backlight 'intel_backlight' level updated: 0.45.
Backlight 'intel_backlight' level updated: 0.40.
Backlight 'intel_backlight' level updated: 0.35.
Backlight 'intel_backlight' level updated: 0.30.
Backlight 'intel_backlight' level updated: 0.25.
Backlight 'intel_backlight' level updated: 0.20.
Backlight 'intel_backlight' level updated: 0.15.

get_backlight(): No such device or address
Failed to get backlight pct.
```

Clight itself is actually working fine, I was just trying to tweak the target brightness levels. Quick capture of the debug log attached anyway in case it's helpful.
[run.log](https://github.com/FedeDP/Clight/files/10687507/run.log)

FedeDP, 2023-03-02:
Hi! Thanks for opening this issue. And sorry for the late feedback!
I will try to reproduce this asap and let you know, in the meantime thanks for using Clight and reporting the issue!

FedeDP, 2023-03-03:
Fixed in master ;) Thank you for reporting this!

commit 09053e4c (2023-03-03): fix(wizard): fix `sd_bus_message_read` to fetch bl interface backlight.

Fixes #272.
closed 2023-03-03 by FedeDP