FedeDP/Clight #275: [QUESTION] How to disable DDC?

jclsn, 2023-03-03:
I would like to keep my monitor from dimming and can't find the setting. Is there a possibility without recompiling?

FedeDP, 2023-03-03:
Hi! At the moment this is not possible; however I was recently thinking about allowing that, since I am refactoring that part of code in Clightd.
So, look forward for the next Clightd release for that :)

FedeDP, 2023-03-03:
This commit: https://github.com/FedeDP/Clightd/pull/92/commits/932ea16209ed10f626ac745a5606e916d3c1f04b fill fix the issue :)
You will have a way (using an env variable) to disable Clightd DDC support.

jclsn, 2023-03-10:
Great, can you make it possible from the config? That would be a bit more convenient imo

FedeDP, 2023-03-10:
Well i can't add that to the Clight config because there is no way to set an env variable to an already running process.
The systemd unit for Clightd will be easily customizable though; i will make sure to update Clightd wiki pages (and add an entry in the Clight FAQ) once the PR is merged!

jclsn, 2023-03-10:
Okay, just modify the service file then. Will try that when it bugs me the next time. I never got why you are shipping clight in two packages anyway. I mean, the daemon could just get installed along with with clight

FedeDP, 2023-03-12:
> I never got why you are shipping clight in two packages anyway. I mean, the daemon could just get installed along with with clight

Yep, that would work fine, and perhaps it would be easier to ship it.
At the same time:
1. I don't want Clightd to be completely tied to Clight (everyone can write an user service that uses Clightd as a backend, and i don't want to stop anyone from doing that). Of course, you could do it even if Clightd was bundled with Clight, but then it would tie them together too much imo
2. I don't want their release cycles to be tied; instead, i want to be able to focus on eg: Clight for a couple of months, and then release it, without caring about Clightd.  Nowadays, i have very limited time to work on my side projects therefore it is much easier for me to manage them in this way.
Of course, sometimes a new Clight feat depends on a new Clightd feat, and then their development cycle is tied, but i can't do much about that :D

jclsn, 2023-03-12:
Okay, yeah organizing the code well is a good idea. From that perspective it makes sense

FedeDP, 2023-06-07:
Clightd PR Is now merged :) and clightd wiki Page updated: https://github.com/FedeDP/Clightd/wiki/Backlight.  I think we can close this one.
The environment variable is named "CLIGHTD_BL_DDC_ENABLED"!

closed 2023-06-07 by FedeDP