FedeDP/Clight #276: [BUG] Compilation issue: 'libmodule' not found

sphh, 2023-03-20:
_Please read on, I have the [`libmodule`](https://github.com/FedeDP/libmodule) compiled and installed as described there._

**Clight version the issue has been seen with:**
> Latest github (https://github.com/FedeDP/Clight/commit/da72508f6fd0587e68ef4c92bbdd439650f0d2b2)

**Used distribution:**
> Linux Mint 21.1 Cinnamon

**Describe the bug**
I followed the instructions in https://github.com/FedeDP/libmodule#user-content-build-dep-and-how-to-build and https://github.com/FedeDP/libmodule#user-content-installation---debian to build and install `libmodule`.

I then cloned the github repository and followed https://github.com/FedeDP/Clight/wiki/Build#user-content-how-to-build. I get the following error:
```
$ cmake     -G "Unix Makefiles"     -DCMAKE_INSTALL_PREFIX=/usr     -DCMAKE_INSTALL_LIBDIR=lib     -DCMAKE_BUILD_TYPE="Release"     ..
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for modules 'popt;gsl;libconfig;libmodule>=5.0.0'
--   No package 'libmodule' found
CMake Error at /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:603 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:825 (_pkg_check_modules_internal)
  CMakeLists.txt:44 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/home/sph/Software/github/Clight/build/CMakeFiles/CMakeOutput.log".
```

**Expected behavior**
`clight` should compile

**To Reproduce**
See bug description.

**Additional information**
```
$ dpkg -L libmodule
/usr
/usr/include
/usr/include/module
/usr/include/module/cmn.h
/usr/include/module/ctx.h
/usr/include/module/mem
/usr/include/module/mem/mem.h
/usr/include/module/mod.h
/usr/include/module/mod_easy.h
/usr/include/module/structs
/usr/include/module/structs/bst.h
/usr/include/module/structs/itr.h
/usr/include/module/structs/list.h
/usr/include/module/structs/map.h
/usr/include/module/structs/queue.h
/usr/include/module/structs/stack.h
/usr/include/module/thpool
/usr/include/module/thpool/thpool.h
/usr/lib
/usr/lib/libmodule_core.so.6.0.0
/usr/lib/libmodule_mem.so.6.0.0
/usr/lib/libmodule_structs.so.6.0.0
/usr/lib/libmodule_thpool.so.6.0.0
/usr/lib/libmodule_utils_internal.so.6.0.0
/usr/share
/usr/share/licenses
/usr/share/licenses/libmodule
/usr/share/licenses/libmodule/LICENSE
/usr/share/pkgconfig
/usr/share/pkgconfig/libmodule_core.pc
/usr/share/pkgconfig/libmodule_mem.pc
/usr/share/pkgconfig/libmodule_structs.pc
/usr/share/pkgconfig/libmodule_thpool.pc
/usr/lib/libmodule_core.so
/usr/lib/libmodule_core.so.6
/usr/lib/libmodule_mem.so
/usr/lib/libmodule_mem.so.6
/usr/lib/libmodule_structs.so
/usr/lib/libmodule_structs.so.6
/usr/lib/libmodule_thpool.so
/usr/lib/libmodule_thpool.so.6
/usr/lib/libmodule_utils_internal.so
/usr/lib/libmodule_utils_internal.so.6
```

I cannot spot a `libmodule.so.6` there. Is this the problem? Should this bug be moved to [`libmodule`](https://github.com/FedeDP/libmodule)?

FedeDP, 2023-03-20:
Hi! Thanks for opening this issue!
It seems like you built libmodule from master. Clight is not compatible with libmodule master; you should stick with libmodule latest tag!

FedeDP, 2023-03-23:
> Maybe this should be mentioned in the README?

You are right, i definitely should! Mind to open a PR?
Otherwise i'll do it as soon as i got some spare time :)

sphh, 2023-03-23:
I guess you might be quicker with the change than me. And if not, it will be at least consistent.

The other thing I missed is that I also have to compile Clightd. Something like this at the start of section ‘How to build’ would have saved me some time:

1. Build and install [`libmodule`](https://github.com/FedeDP/libmodule) after checking out it latest tag (e.g. before building do a `git checkout 5.0.1`) _**or**_ install its package.
2. Build and install [`Clightd`](https://github.com/FedeDP/Clightd/wiki/Build)
3. Build and install `Clight` as follows.

FedeDP, 2023-03-23:
I added a couple of **NOTE**s in the wiki page.

However, please note that libmodule and Clightd deps were actually already listed between the dependencies section: https://github.com/FedeDP/Clight/wiki/Build#dependencies

I hope that now it is clearer! Thank you for your suggestions!

sphh, 2023-03-23:
You are absolutely right: Both `libmodule` and `Clightd` were mentioned, but it was not easy to spot (at least for me).

I like the **NOTE**s. That's A Good Idea and it makes it so much clearer. Thanks!