FedeDP/Clight #282: [BUG] ALS Not Work

KairuiLiu, 2023-08-11:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> Clightd found, version: 5.8-8a90e9d.

**Used distribution:**
> AUR?

**Describe the bug**

ASL can found on my laptop

```
Udevadm info starts with the device specified by the devpath and then
walks up the chain of parent devices. It prints for every device
found, all possible attributes in the udev rules key format.
A rule to match, can be composed by the attributes of the device
and the attributes from one single parent device.

  looking at device '/devices/pci0000:00/0000:00:12.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0007/HID-SENSOR-200041.3.auto/iio:device0':
    KERNEL=="iio:device0"
    SUBSYSTEM=="iio"
    DRIVER==""
    ATTR{buffer/data_available}=="0"
    ATTR{buffer/direction}=="in"
    ATTR{buffer/enable}=="0"
    ATTR{buffer/hwfifo_enabled}=="0"
    ATTR{buffer/hwfifo_timeout}=="0.000000"
    ATTR{buffer/length}=="2"
    ATTR{buffer/watermark}=="1"
    ATTR{buffer0/data_available}=="0"
    ATTR{buffer0/direction}=="in"
    ATTR{buffer0/enable}=="0"
    ATTR{buffer0/hwfifo_enabled}=="0"
    ATTR{buffer0/hwfifo_timeout}=="0.000000"
    ATTR{buffer0/in_illuminance_en}=="0"
    ATTR{buffer0/in_illuminance_index}=="1"
    ATTR{buffer0/in_illuminance_type}=="le:s32/32>>0"
    ATTR{buffer0/in_intensity_both_en}=="0"
    ATTR{buffer0/in_intensity_both_index}=="0"
    ATTR{buffer0/in_intensity_both_type}=="le:s32/32>>0"
    ATTR{buffer0/in_timestamp_en}=="0"
    ATTR{buffer0/in_timestamp_index}=="2"
    ATTR{buffer0/in_timestamp_type}=="le:s64/64>>0"
    ATTR{buffer0/length}=="2"
    ATTR{buffer0/watermark}=="1"
    ATTR{current_timestamp_clock}=="realtime"
    ATTR{in_illuminance_hysteresis}=="0.001000"
    ATTR{in_illuminance_hysteresis_relative}=="0.010000"
    ATTR{in_illuminance_offset}=="0"
    ATTR{in_illuminance_raw}=="2350"
    ATTR{in_illuminance_sampling_frequency}=="10.000000"
    ATTR{in_illuminance_scale}=="0.001000000"
    ATTR{in_intensity_both_raw}=="2350"
    ATTR{in_intensity_hysteresis}=="0.001000"
    ATTR{in_intensity_hysteresis_relative}=="0.010000"
    ATTR{in_intensity_offset}=="0"
    ATTR{in_intensity_sampling_frequency}=="10.000000"
    ATTR{in_intensity_scale}=="0.001000000"
    ATTR{name}=="als"
    ATTR{power/control}=="auto"
    ATTR{power/runtime_active_time}=="0"
    ATTR{power/runtime_status}=="unsupported"
    ATTR{power/runtime_suspended_time}=="0"
    ATTR{scan_elements/in_illuminance_en}=="0"
    ATTR{scan_elements/in_illuminance_index}=="1"
    ATTR{scan_elements/in_illuminance_type}=="le:s32/32>>0"
    ATTR{scan_elements/in_intensity_both_en}=="0"
    ATTR{scan_elements/in_intensity_both_index}=="0"
    ATTR{scan_elements/in_intensity_both_type}=="le:s32/32>>0"
    ATTR{scan_elements/in_timestamp_en}=="0"
    ATTR{scan_elements/in_timestamp_index}=="2"
    ATTR{scan_elements/in_timestamp_type}=="le:s64/64>>0"
    ATTR{trigger/current_trigger}=="als-dev0"

  looking at parent device '/devices/pci0000:00/0000:00:12.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0007/HID-SENSOR-200041.3.auto':
    KERNELS=="HID-SENSOR-200041.3.auto"
    SUBSYSTEMS=="platform"
    DRIVERS=="hid_sensor_als"
    ATTRS{driver_override}=="(null)"
    ATTRS{power/autosuspend_delay_ms}=="3000"
    ATTRS{power/control}=="auto"
    ATTRS{power/runtime_active_time}=="90973"
    ATTRS{power/runtime_status}=="active"
    ATTRS{power/runtime_suspended_time}=="1419447"

  looking at parent device '/devices/pci0000:00/0000:00:12.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8087:0AC2.0007':
    KERNELS=="001F:8087:0AC2.0007"
    SUBSYSTEMS=="hid"
    DRIVERS=="hid-sensor-hub"
    ATTRS{power/control}=="auto"
    ATTRS{power/runtime_active_time}=="0"
    ATTRS{power/runtime_status}=="unsupported"
    ATTRS{power/runtime_suspended_time}=="0"

  looking at parent device '/devices/pci0000:00/0000:00:12.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}':
    KERNELS=="{33AECD58-B679-4E54-9BD9-A04D34F0C226}"
    SUBSYSTEMS=="ishtp"
    DRIVERS=="ish-hid"
    ATTRS{power/control}=="auto"
    ATTRS{power/runtime_active_time}=="0"
    ATTRS{power/runtime_status}=="unsupported"
    ATTRS{power/runtime_suspended_time}=="0"

  looking at parent device '/devices/pci0000:00/0000:00:12.0':
    KERNELS=="0000:00:12.0"
    SUBSYSTEMS=="pci"
    DRIVERS=="intel_ish_ipc"
    ATTRS{ari_enabled}=="0"
    ATTRS{broken_parity_status}=="0"
    ATTRS{class}=="0x070000"
    ATTRS{consistent_dma_mask_bits}=="32"
    ATTRS{d3cold_allowed}=="1"
    ATTRS{device}=="0x51fc"
    ATTRS{dma_mask_bits}=="32"
    ATTRS{driver_override}=="(null)"
    ATTRS{enable}=="1"
    ATTRS{index}=="8"
    ATTRS{irq}=="26"
    ATTRS{label}=="Onboard - Other"
    ATTRS{local_cpulist}=="0-19"
    ATTRS{local_cpus}=="fffff"
    ATTRS{msi_bus}=="1"
    ATTRS{numa_node}=="-1"
    ATTRS{power/control}=="on"
    ATTRS{power/runtime_active_time}=="1627412"
    ATTRS{power/runtime_status}=="active"
    ATTRS{power/runtime_suspended_time}=="0"
    ATTRS{power/wakeup}=="disabled"
    ATTRS{power/wakeup_abort_count}==""
    ATTRS{power/wakeup_active}==""
    ATTRS{power/wakeup_active_count}==""
    ATTRS{power/wakeup_count}==""
    ATTRS{power/wakeup_expire_count}==""
    ATTRS{power/wakeup_last_time_ms}==""
    ATTRS{power/wakeup_max_time_ms}==""
    ATTRS{power/wakeup_total_time_ms}==""
    ATTRS{power_state}=="D0"
    ATTRS{remove}=="(not readable)"
    ATTRS{rescan}=="(not readable)"
    ATTRS{resource0}=="(not readable)"
    ATTRS{revision}=="0x01"
    ATTRS{subsystem_device}=="0x1693"
    ATTRS{subsystem_vendor}=="0x1043"
    ATTRS{vendor}=="0x8086"

  looking at parent device '/devices/pci0000:00':
    KERNELS=="pci0000:00"
    SUBSYSTEMS==""
    DRIVERS==""
    ATTRS{power/control}=="auto"
    ATTRS{power/runtime_active_time}=="0"
    ATTRS{power/runtime_status}=="unsupported"
    ATTRS{power/runtime_suspended_time}=="0"
    ATTRS{waiting_for_supplier}=="0"
```

and read data by system

```
# cover ALS
cat /sys/bus/iio/devices/iio\:device0/in_illuminance_raw
348
# remove cover
cat /sys/bus/iio/devices/iio\:device0/in_illuminance_raw
156942
```

and data can be read by clight?

```
# cover the ALS
busctl call org.clightd.clightd /org/clightd/clightd/Sensor/Als org.clightd.clightd.Sensor Capture "sis" "" 5 ""
sad "/dev/iio:device0" 5 0 0 0 0 0
# remove the cover
busctl call org.clightd.clightd /org/clightd/clightd/Sensor/Als org.clightd.clightd.Sensor Capture "sis" "" 5 ""
sad "/dev/iio:device0" 5 0.431044 0.431044 0.431044 0.431044 0.431044
```

but do nothing when light change

```
clight -d /dev/iio:device0
Using deprecated /etc/default/clight.conf config file whose support will be dropped in the near future. Please move to new one at: /etc/clight/clight.conf.
Config file: file I/O error at line 0.
Clightd found, version: 5.8-8a90e9d.
org.freedesktop.ScreenSaver dbus interface exposed.
AC cable connected.
Laptop lid opened.
Next alarm due to: Sat Aug 12 07:41:00 2023
Ambient brightness: 0.307, Screen brightness: 0.490 -> Screen backlight: 0.483.   <-------------- I put my flash on ALS
Ambient brightness: 0.307, Screen brightness: 0.494 -> Screen backlight: 0.481.
Ambient brightness: 0.307, Screen brightness: 0.494 -> Screen backlight: 0.481.
Ambient brightness: 0.307, Screen brightness: 0.494 -> Screen backlight: 0.481.  
Ambient brightness: 0.307, Screen brightness: 0.459 -> Screen backlight: 0.495.  <-------------- remove flash, cover the ALS
Ambient brightness: 0.307, Screen brightness: 0.494 -> Screen backlight: 0.481.
Ambient brightness: 0.307, Screen brightness: 0.494 -> Screen backlight: 0.481.
```

FedeDP, 2023-08-11:
Hi! Thanks for opening this bug report!
Can you enable Clight verbose logs? They should log from which device captures are being taken!
Thank you very much!

KairuiLiu, 2023-08-12:
Yes, in the following log, I run clight about 3 mins, in the last 2 mins, I cover the ALS, but the Ambient brightness didn't change

```
Using deprecated /etc/default/clight.conf config file whose support will be dropped in the near future. Please move to new one at: /etc/clight/clight.conf.
Config file: file I/O error at line 0.
Clightd found, version: 5.8-8a90e9d.
Disarmed timerfd on fd 9.
AC screen backlight curve: y = -0.024825 + 0.191641x + -0.008928x^2
BL
^
|        **
|      **  
|     *    
|    *     
|   *      
|  *       
|          
| *        
|*         
|          
*---------->BR
BATT screen backlight curve: y = -0.010629 + 0.153844x + -0.007284x^2
BL
^
|          
|          
|       ***
|     **   
|    *     
|   *      
|  *       
| *        
|          
|*         
*---------->BR
org.freedesktop.ScreenSaver dbus interface exposed.
Disarmed timerfd on fd 31.
AC cable connected.
Laptop lid opened.
Emitting 'AcState' property
Set timeout of 5s 0ns on fd 34.
Emitting 'LidState' property
Set timeout of 0s 1ns on fd 35.
Emitting 'ScreenBr' property
Currently inside an event.
Next alarm due to: Sat Aug 12 18:12:00 2023
Set timeout of 1766s 0ns on fd 35.
Emitting 'Sunrise' property
Emitting 'Sunset' property
Emitting 'DayTime' property
Emitting 'NextEvent' property
Emitting 'InEvent' property
Set timeout of 0s 1ns on fd 36.
Sensor '/dev/iio:device0' is now available.
Resumed as a sensor is now available.
Emitting 'SensorAvail' property
Captured [5/5] from '/dev/iio:device0'. Ambient brightness: 0.285.
Set timeout of 300s 0ns on fd 36.
Emitting 'AmbientBr' property
Emitting 'ScreenBr' property
Content calib: wmax: 0.649, wmin: 0.249, new_bl: 0.327
Ambient brightness: 0.285, Screen brightness: 0.804 -> Screen backlight: 0.327.
Emitting 'BlPct' property
Backlight 'intel_backlight' level updated: 0.36.
Emitting 'BlPct' property
Backlight 'intel_backlight' level updated: 0.33.
Emitting 'BlPct' property
Set timeout of 5s 0ns on fd 34.
Content calib: wmax: 0.649, wmin: 0.249, new_bl: 0.453
Ambient brightness: 0.285, Screen brightness: 0.490 -> Screen backlight: 0.453.
[... 314 lines cut ...]
Backlight 'intel_backlight' level updated: 0.45.
Emitting 'BlPct' property
Backlight 'intel_backlight' level updated: 0.47.
Emitting 'BlPct' property
Set timeout of 5s 0ns on fd 34.
Content calib: wmax: 0.649, wmin: 0.249, new_bl: 0.473
Ambient brightness: 0.285, Screen brightness: 0.439 -> Screen backlight: 0.473.
Emitting 'ScreenBr' property
Emitting 'BlPct' property
Backlight 'intel_backlight' level updated: 0.47.
Emitting 'BlPct' property
Backlight 'intel_backlight' level updated: 0.42.
Emitting 'BlPct' property
Set timeout of 5s 0ns on fd 34.
Content calib: wmax: 0.649, wmin: 0.249, new_bl: 0.473
Ambient brightness: 0.285, Screen brightness: 0.439 -> Screen backlight: 0.473.
Emitting 'ScreenBr' property
Emitting 'BlPct' property
Backlight 'intel_backlight' level updated: 0.47.
Emitting 'BlPct' property
Backlight 'intel_backlight' level updated: 0.47.
Emitting 'BlPct' property
Received 2. Leaving.

```

FedeDP, 2023-08-17:
Thanks! I only see a single capture being taken in the output above:
> Captured [5/5] from '/dev/iio:device0'. Ambient brightness: 0.285.

Can you share the log file too? It has conf options logged and datetime info for each line!

FedeDP, 2023-08-17:
> Yes, in the following log, I run clight about 3 mins, in the last 2 mins, I cover the ALS, but the Ambient brightness didn't change

I see now! Clight takes captures every 10mins by default! So, you need to wait or you can decrease default timeouts from config file!

FedeDP, 2023-10-12:
Closing; feel free to reopen if you have more questions!

closed 2023-10-12 by FedeDP