FedeDP/Clight #283: Windows possible?

sengiv, 2023-10-02:
How do I go about running in windows 10?

I can write a windows .net port, if can't be done now.

Please point me in the right direction 🙏

FedeDP, 2023-10-12:
Well, given that:
* Clight requires a dbus system service (Clightd)
* both Clight and Clightd use `libmodule` that is not available for windows
* both Clight and Clightd use systemd dbus library (`sd-bus`)

I assume it would be very very hard to port Clight to windows, sorry. I always developed it to be linux-specific :) 
Nowadays, libmodule works on osx and bsd too, but since both rely on DBus and its systemd library, i guess it would be pretty hard to port them to osx/bsd too.

FedeDP, 2023-11-04:
Closing this one! Btw thanks for the interest, please let me know if you start working on a porting (or a completely new impl from scratch!).
It would be really interesting to learn the Windows API!

closed 2023-11-04 by FedeDP