FedeDP/Clight #285: Getting errors with new cofig layout[BUG]

hunterzero99, 2023-11-03:
I'm getting some errors with the new config file format.  

I get these errors seemingly no matter what.  I have sidestepped them by copying the new config file to ~/.config/clight.conf
> Using deprecated /etc/default/clight.conf config file whose support will be dropped in the near future. Please move to new one at: /etc/clight/clight.conf.
Config file: file I/O error at line 0.

Then I receive errors like this:
> Config file: cannot open include file at line 30.

Which I have resolved by changing the following lines:

> @include "modules.conf.d/backlight.conf"

to include /etc/clight/ in the directory name.

> @include "/etc/clight/modules.conf.d/inhibit.conf"

FedeDP, 2023-11-04:
Hi! Thanks for opening this issue!
> Config file: file I/O error at line 0.

I am not really sure what's causing this issue; never saw that! Can you share first 5 lines of `/etc/default/clight.conf` file?
> Then I receive errors like this:

Yep, because if you copy it to your local config, it won't find the `include`d config files anymore since they are provided as relative paths, so your fix is correct.
To fix the latter, i think i should move to always use absolute path, so that wherever the configs are, the includes will be fine.

FedeDP, 2023-11-04:
Well, i was wrong: since i used ` config_set_include_dir(&cfg, dirname(config_file_dup));`, it means that each config will resolve its includes relative to its folder.
This is the correct behavior IMHO, so if you copy `/etc/clight/clight.conf` to your local folder, you are expected to also copy interesting configs from `/etc/clight/modules.conf.d/` (ie: the ones you want to override) and then edit the copied `clight.conf` to only include needed configs.

hunterzero99, 2023-11-05:
I actually don't have a file at /etc/default/clight.conf

`[user@home ~]$ cat /etc/default/clight.conf
cat: /etc/default/clight.conf: No such file or directory`

FedeDP, 2023-11-06:
> I actually don't have a file at /etc/default/clight.conf

That actually explains **a lot**!
Indeed there was a bug where the code always tried to parse `/etc/default/clight.conf` even when not present.
It is fixed in latest master.

closed 2023-11-30 by FedeDP