FedeDP/Clight #288: Black Screen - Read FAQ - Can not convert to action

Kerrnel, 2024-01-27:
The FAQ says to change the curves in clight.conf to 0.1 from 0.0 to resolve screen going black on minimum light

```
 $ grep -r 0.0 .
./modules.conf.d/backlight.conf:    # trans_step = 0.05;
./modules.conf.d/backlight.conf:    # trans_fixed = 1000;
./modules.conf.d/dimmer.conf:    # trans_steps = [ 0.05, 0.05 ];
./modules.conf.d/dimmer.conf:    # trans_fixed = [ 1000, 1000 ];
./modules.conf.d/gamma.conf:    # temp = [ 6500, 4000 ];
./modules.conf.d/keyboard.conf:    # ac_regression_points = [ 1.0, 0.97, 0.93, 0.88, 0.81, 0.74, 0.61, 0.45, 0.29, 0.15, 0.0 ];
./modules.conf.d/keyboard.conf:    # batt_regression_points = [ 0.80, 0.78, 0.75, 0.71, 0.65, 0.59, 0.52, 0.36, 0.23, 0.15, 0.0 ];
./modules.conf.d/screen.conf:    ## Set it to 0.0 to disable screen-content based backlight compensation.
./modules.conf.d/sensor.conf:    # ac_regression_points = [ 0.0, 0.15, 0.29, 0.45, 0.61, 0.74, 0.81, 0.88, 0.93, 0.97, 1.0 ];
./modules.conf.d/sensor.conf:    # batt_regression_points = [ 0.0, 0.15, 0.23, 0.36, 0.52, 0.59, 0.65, 0.71, 0.75, 0.78, 0.80 ];
./modules.conf.d/sensor.conf:        # ac_regression_points = [ 0.0, 0.18, 0.22, 0.33, 0.55, 0.64, 0.71, 0.80, 0.90, 0.97, 1.0 ];
./modules.conf.d/sensor.conf:        # batt_regression_points =  [ 0.0, 0.15, 0.29, 0.45, 0.61, 0.74, 0.81, 0.88, 0.93, 0.97, 1.0 ];
./modules.conf.d/sensor.conf:        # ac_regression_points = [ 0.0, 0.18, 0.22, 0.33, 0.55, 0.64, 0.71, 0.80, 0.90, 0.97, 1.0 ];
./modules.conf.d/sensor.conf:        # batt_regression_points =  [ 0.0, 0.15, 0.29, 0.45, 0.61, 0.74, 0.81, 0.88, 0.93, 0.97, 1.0 ];

```
Searched for curve, gamma etc. Could this be more explicit?

FedeDP, 2024-03-09:
Hi! Sorry for the late response, i totally forgot about this one...
So, you need to update `sensor.conf.{ac,batt}_regression_points` minimum value from 0.0 to something else. i will update the FAQs!

FedeDP, 2024-03-09:
Here it is! https://github.com/FedeDP/Clight/wiki/FAQ#clight-sets-0-backlight-level-but-my-screen-turns-black

closed 2024-03-09 by FedeDP