FedeDP/Clight #289: [BUG] move clight.conf to /etc/clight/ ? A better warning

promeneur, 2024-01-28:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> ...4.10

**Used distribution:**
> ...openSUSE Tumbleweed

It is said
(W)[11:18:58]{config.c:414}	Using deprecated /etc/default/clight.conf config file whose support will be dropped in the near future. Please move to new one at: /etc/clight/clight.conf.

But in /etc/default/ there is no clight.conf but clight.conf.rpmsave.
and in this file all is commented.

In  /etc/clight/ there is a clight.conf but very diiferent from customized clight.conf.rpmsave.
And if i want to put in the statement "devname = "HD Pro Webcam C920";"  I don't know where to put in.

Happily, i found sensors.conf where to put in my statement about the webcam to use.

I think a better warning is needed : 

/etc/default/clight.conf is deprecated.
clight.conf becomes several conf files. Each chapter of the clight.conf is now one conf file with the same name in /etc/clight/modules.conf.d/
Please move your customized statements to the corresponding files.

Thanks

FedeDP, 2024-02-06:
Hi! Thanks for opening this bug report; similar to #290 , this is already fixed in master: https://github.com/FedeDP/Clight/commit/0d6d075e027cf6a802fa7c837264823f69bdd34e

FedeDP, 2024-03-09:
Should be fixed by the just released Clight 4.11!

closed 2024-03-09 by FedeDP