FedeDP/Clight #291: [BUG] many, many Ambient brightness: 0.836, Screen brightness: 0.129 -> Screen backlight: 1.000.

promeneur, 2024-01-28:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> ...4.10

**Used distribution:**
> ...openSUSE Tumbleweed

The monitor is plugged to the PC, 

using HDMI wire.

**Describe the bug**

In clight.log, I never seen such messages : 

(I)[11:19:03]{backlight.c:522}	Ambient brightness: 0.836, Screen brightness: 0.129 -> Screen backlight: 1.000.

[clight.log](https://github.com/FedeDP/Clight/files/14075741/clight.log)

FedeDP, 2024-02-06:
Hi! That's because since Clight 4.9, SCREEN module has been completely reworked and it does now capture screenshots every few seconds to compute correct backlight-adjusted value given screen-emitted brightness.  
Note that the messages should only be printed whenever the screen backlight changes, not always.
If the spam is too much, you can disable SCREEN module. I might consider a "notification cooldown" period workaround too perhaps, even if i wouldn't really like the added complexity.

promeneur, 2024-02-06:
OK thanks

FedeDP, 2024-03-09:
I'll close this one since it is not a real issue after all. Let me know if you'd prefer a different fix (ie: a cool-down period between notifications; as i said, i would not love the added complexity and the fact that the log file would not log everything Clight is doing though.

closed 2024-03-09 by FedeDP