FedeDP/Clight #293: [BUG] Clight changes the backlight level twice every capture

3cky, 2024-02-15:
**Clight version the issue has been seen with:**
Clight 4.11-6a5ad96 / Clightd 5.9-7d447d0 / Fedora 39

**Used distribution:**
Source build.

**Describe the bug**
Clight changes the backlight level twice on every capture, one time with the calculated value and the second time with the other (wrong) value:

```
Sensor '/etc/clightd/sensors.d/als' is now available.
Resumed as a sensor is now available.
Emitting 'NextEvent' property
Emitting 'SensorAvail' property
Captured [1/1] from '/etc/clightd/sensors.d/als'. Ambient brightness: 0.488.
Ambient brightness: 0.488 -> Screen backlight: 0.698.
Set timeout of 10s 0ns on fd 42.
Emitting 'AmbientBr' property
Emitting 'BlPct' property
Backlight 'intel_backlight' level updated: 0.70.
Emitting 'BlPct' property
Backlight '/dev/i2c-13' level updated: 0.70.
Backlight 'intel_backlight' level updated: 0.65.
Emitting 'BlPct' property
Backlight '/dev/i2c-13' level updated: 0.65.
Captured [1/1] from '/etc/clightd/sensors.d/als'. Ambient brightness: 0.488.
Ambient brightness: 0.488 -> Screen backlight: 0.698.
Set timeout of 10s 0ns on fd 42.
Emitting 'AmbientBr' property
Emitting 'BlPct' property
Backlight 'intel_backlight' level updated: 0.70.
Emitting 'BlPct' property
Backlight '/dev/i2c-13' level updated: 0.70.
Backlight 'intel_backlight' level updated: 0.65.
Emitting 'BlPct' property
Backlight '/dev/i2c-13' level updated: 0.65.
Captured [1/1] from '/etc/clightd/sensors.d/als'. Ambient brightness: 0.488.
Ambient brightness: 0.488 -> Screen backlight: 0.698.
Set timeout of 10s 0ns on fd 42.
Emitting 'AmbientBr' property
Emitting 'BlPct' property
Backlight 'intel_backlight' level updated: 0.70.
Emitting 'BlPct' property
Backlight '/dev/i2c-13' level updated: 0.70.
Backlight 'intel_backlight' level updated: 0.65.
Emitting 'BlPct' property
Backlight '/dev/i2c-13' level updated: 0.65.
```

**Expected behavior**
The backlight level should change once every capture.

**To Reproduce**
`clight.conf`:
```
verbose = true;
backlight :
{
  ac_timeouts = [ 10, 10, 10 ];
};
keyboard :
{
  disabled = true;
};
sensor:
{
  captures = [ 1, 1 ];
  devname = "als";
};
```
```
$ cat /etc/clightd/sensors.d/als
2000
```

3cky, 2024-02-16:
Did some investigation and it looks like it's not a Clight issue but Clightd. I bisected it and it seems that this bug was introduced by [this commit](https://github.com/FedeDP/Clightd/commit/b762958e19d96e42ba49be856519e4c29e3e06d2).

FedeDP, 2024-02-16:
Hi! Thanks for reporting this issue and thanks for the investigation! Are you willing to try to fix it? Otherwise I can dig it further and hopefully find a solution:)

3cky, 2024-02-17:
Hi, I took a look into the Clightd code and the problem seems to be more complex. While [ce65493](https://github.com/FedeDP/Clightd/commit/ce65493b57333abdac70d0289ad3954addd17662) is the last correctly working code, commit [b762958](https://github.com/FedeDP/Clightd/commit/b762958e19d96e42ba49be856519e4c29e3e06d2) is not the root of the issue, at least not the single root, because it was effectively reverted in commit [65bd59c](https://github.com/FedeDP/Clightd/commit/65bd59c55a2dd2a05054e73a311b20403674f449). That's all I dug for now, I will report here if I find something more.