FedeDP/Clight #294: No smooth transition. Settings are ignored. [BUG]

sonofevil, 2024-03-08:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> 4.10-1

**Used distribution:**
> Arch Linux

**Describe the bug**
Clight abruptly changes backlight brightness in steps of 3-7% which is very jarring. When I start clight, it tells me `Using deprecated /etc/default/clight.conf config file whose support will be dropped in the near future. Please move to new one at: /etc/clight/clight.conf.` which is absurd because there is no '/etc/default/clight.conf' file, but there is a '/etc/clight/clight.conf'. Thus when I run 'clight' it tells me 'Config file: file I/O error at line 0.' So I created '/etc/default/clight.conf' with '@include "/etc/clight/clight.conf"', then changed all the paths in '/etc/default/clight.conf' to absolute paths so it'd would work. Now at least there is no error. Still the config is seemingly ignored though. I get messages about DPMS, Keyboard backlight, screensaver, daytime etc, even though I have all of these modules disabled in the respective configs. I also have in backlight.conf:

```
    trans_step = 0.002;

    ## Transition timeout in ms
     trans_timeout = 30;
    
    ## When > 0, use a fixed transition duration (in ms),
    ## overriding trans_timeout and trans_step configs.
     trans_fixed = 1000;
```

But no, it still jumps in ~5% steps.

**Expected behavior**
I need transitions to be so smooth that I barely notice them.

**To Reproduce**
No idea.

```
(W)[22:57:49]{config.c:414}	Using deprecated /etc/default/clight.conf config file whose support will be dropped in the near future. Please move to new one at: /etc/clight/clight.conf.
(W)[22:57:49]{opts.c:438}	DPMS_CONF: wrong AC 'timeout' value (<= dimmer timeout). Resetting default value.
(W)[22:57:49]{opts.c:443}	DPMS_CONF: wrong BATT 'timeout' value (<= dimmer timeout). Resetting default value.

Clight
* Software version:		4.10-f422180
* Global config dir:		/etc/clight
* Global data dir:		/usr/share/clight
* Starting time:		Fri Mar  8 22:57:49 2024

Starting options:

### GENERIC ###
* Verbose (debug):		Disabled
* ResumeDelay:		0

### BACKLIGHT ###
* Smooth trans:		Enabled
* Smooth step:		0.05
* Smooth timeout:		500
* Smooth fixed:		0
* Daily timeouts:		AC 600	BATT 1200
* Nightly timeouts:		AC 2700	BATT 5400
* Event timeouts:		AC 300	BATT 600
* Shutter threshold:		0.00
* Autocalibration:		Enabled
* Pause on lid closed:		Disabled
* Capture on lid opened:		Disabled
* Restore On Exit:		Disabled
* Delay on hotplug:		0

### SENSOR ###
* Captures:		AC 5	BATT 5
* Device:		Unset
* Settings:		Unset

### KEYBOARD ###
* Timeouts:		AC 15	BATT 5

### GAMMA ###
* Smooth trans:		Enabled
* Smooth steps:		50
* Smooth timeout:		300
* Daily screen temp:		6500
* Nightly screen temp:		4000
* Long transition:		Enabled
* Ambient gamma:		Enabled
* Restore On Exit:		Disabled

### DAYTIME ###
* User position:		Unset
* User set sunrise:		Unset
* User set sunset:		Unset
* Event duration:		1800
* Sunrise offset:		0
* Sunset offset:		0

### DIMMER ###
* ENTER Smooth trans:		Enabled
* ENTER Smooth step:		0.05
* ENTER Smooth timeout:		10000
* ENTER Smooth fixed:		0
* EXIT Smooth trans:		Enabled
* EXIT Smooth step:		0.05
* EXIT Smooth timeout:		10000
* EXIT Smooth fixed:		0
* Timeouts:		AC 9999	BATT 9999
* Backlight pct:		0.01

### DPMS ###
* Timeouts:		AC 900	BATT 300

### SCREEN ###
* Timeouts:		AC 30	BATT -1

### INHIBIT ###
* Docked:		Disabled
* PowerManagement:		Enabled
* Backlight:		Disabled

(I)[22:57:49]{main.c:145}	Clightd found, version: 5.8-77b20b9.
(I)[22:57:49]{keyboard.c:159}	Keyboard backlight calibration unsupported.
(I)[22:57:49]{interface.c:176}	Monitoring requests to org.freedesktop.ScreenSaver name owner.
(I)[22:57:49]{upower.c:55}	AC cable connected.
(I)[22:57:49]{upower.c:66}	Laptop lid opened.
(W)[22:58:19]{location.c:90}	Timed out waiting on location provided by Geoclue2. Killing module.
(W)[22:58:19]{daytime.c:72}	No location provider found; fallback to DAY daytime.
(W)[22:58:19]{gamma.c:93}	Killing GAMMA as no location provider is available.
(I)[22:58:20]{backlight.c:522}	Ambient brightness: 0.163, Screen brightness: 0.165 -> Screen backlight: 0.346.
(I)[22:58:20]{backlight.c:522}	Ambient brightness: 0.163, Screen brightness: 0.165 -> Screen backlight: 0.346.
(W)[22:58:20]{bus.c:199}	Error in async req: unknown
(I)[22:58:50]{backlight.c:522}	Ambient brightness: 0.163, Screen brightness: 0.153 -> Screen backlight: 0.351.
(I)[22:59:20]{backlight.c:522}	Ambient brightness: 0.163, Screen brightness: 0.157 -> Screen backlight: 0.349.
(I)[22:59:50]{backlight.c:522}	Ambient brightness: 0.163, Screen brightness: 0.157 -> Screen backlight: 0.349.
(I)[23:00:20]{backlight.c:522}	Ambient brightness: 0.163, Screen brightness: 0.153 -> Screen backlight: 0.351.
(I)[23:00:50]{backlight.c:522}	Ambient brightness: 0.163, Screen brightness: 0.173 -> Screen backlight: 0.343.
(I)[23:01:20]{backlight.c:522}	Ambient brightness: 0.163, Screen brightness: 0.176 -> Screen backlight: 0.341.

```

FedeDP, 2024-03-08:
Hi! There were some bugs in Clight 4.10 related to old config management; can you try to upgrade to Clight-git? It should solve those issues!
Also, I need to find the spare time to release Clight 4.11...

FedeDP, 2024-03-09:
Hey, i finally took the time to tag Clight 4.11; it should fix your issues; let me know please!

sonofevil, 2024-03-11:
> Hey, i finally took the time to tag Clight 4.11; it should fix your issues; let me know please!

Thank you, I installed `clight-git 4.11.r1.ge256faa-1` and There is no longer any errors about the config, but the transitions are still not smooth and my transition settings are still ignored.

FedeDP, 2024-03-12:
Mmmh interesting! Can you enable verbose logs?
Moreover, 
> Clight abruptly changes backlight brightness in steps of 3-7% which is very jarring. 

This is most probably caused by `SCREEN` module doing its work; does the same happen when you are on battery (ie: when screen module is disabled)?

Also
> * Smooth trans:		Enabled
* Smooth step:		0.05
* Smooth timeout:		500
* Smooth fixed:		0

It seems like you have smoothing enabled with a 5% step, therefore it is the expected behavior that backlight is updated with 5% steps.

sonofevil, 2024-03-12:
> It seems like you have smoothing enabled with a 5% step, therefore it is the expected behavior that backlight is updated with 5% steps.

So the log says, but my actual config looks like this:

```
    ## Uncomment to disable smooth transitions
    # no_smooth_transition = true;

    ## Transition step in percentage
     trans_step = 0.002;

    ## Transition timeout in ms
     trans_timeout = 1000;
    
    ## When > 0, use a fixed transition duration (in ms),
    ## overriding trans_timeout and trans_step configs.
     trans_fixed = 1000;
```

FedeDP, 2024-03-12:
Is this from `backlight.conf` config file?
Can you share your `/etc/clight/clight.conf` and `/etc/clight/modules.conf.d/backlight.conf`?

sonofevil, 2024-03-12:
Extension changed to make shareable:

[backlight.conf.txt](https://github.com/FedeDP/Clight/files/14572390/backlight.conf.txt)
[clight.conf.txt](https://github.com/FedeDP/Clight/files/14572391/clight.conf.txt)

I tried setting verbose in `clight.conf`, but this too gets ignored. Log from `--verbose`:

```
(D)[13:48:12]{config.c:413}	Reading config file: /home/lutra/.config/clight.conf
(W)[13:48:12]{opts.c:438}	DPMS_CONF: wrong AC 'timeout' value (<= dimmer timeout). Resetting default value.
(W)[13:48:12]{opts.c:443}	DPMS_CONF: wrong BATT 'timeout' value (<= dimmer timeout). Resetting default value.

Clight
* Software version:		4.12-e256faa
* Global config dir:		/etc/clight
* Global data dir:		/usr/share/clight
* Starting time:		Tue Mar 12 13:48:12 2024

Starting options:

### GENERIC ###
* Verbose (debug):		Enabled
* ResumeDelay:		0

### BACKLIGHT ###
* Smooth trans:		Enabled
* Smooth step:		0.05
* Smooth timeout:		500
* Smooth fixed:		0
* Daily timeouts:		AC 600	BATT 1200
* Nightly timeouts:		AC 2700	BATT 5400
* Event timeouts:		AC 300	BATT 600
* Shutter threshold:		0.00
* Autocalibration:		Enabled
* Pause on lid closed:		Disabled
* Capture on lid opened:		Disabled
* Restore On Exit:		Disabled
* Delay on hotplug:		0

### SENSOR ###
* Captures:		AC 5	BATT 5
* Device:		Unset
* Settings:		Unset

### KEYBOARD ###
* Timeouts:		AC 15	BATT 5

### GAMMA ###
* Smooth trans:		Enabled
* Smooth steps:		50
* Smooth timeout:		300
* Daily screen temp:		6500
* Nightly screen temp:		4000
* Long transition:		Enabled
* Ambient gamma:		Enabled
* Restore On Exit:		Disabled

### DAYTIME ###
* User position:		Unset
* User set sunrise:		Unset
* User set sunset:		Unset
* Event duration:		1800
* Sunrise offset:		0
* Sunset offset:		0

### DIMMER ###
* ENTER Smooth trans:		Enabled
* ENTER Smooth step:		0.05
* ENTER Smooth timeout:		10000
* ENTER Smooth fixed:		0
[... 102 lines cut ...]
(D)[13:48:47]{interface.c:230}	Emitting 'BlPct' property
(D)[13:48:48]{backlight.c:736}	Backlight 'intel_backlight' level updated: 0.73.
(D)[13:48:48]{interface.c:230}	Emitting 'BlPct' property
(D)[13:48:48]{backlight.c:736}	Backlight 'intel_backlight' level updated: 0.78.
(D)[13:48:48]{interface.c:230}	Emitting 'BlPct' property
(D)[13:48:49]{backlight.c:736}	Backlight 'intel_backlight' level updated: 0.83.
(D)[13:48:49]{interface.c:230}	Emitting 'BlPct' property
(D)[13:48:49]{backlight.c:736}	Backlight 'intel_backlight' level updated: 0.88.
(D)[13:48:49]{interface.c:230}	Emitting 'BlPct' property
(D)[13:48:50]{backlight.c:736}	Backlight 'intel_backlight' level updated: 0.93.
(D)[13:48:50]{interface.c:230}	Emitting 'BlPct' property
(D)[13:48:50]{backlight.c:736}	Backlight 'intel_backlight' level updated: 0.98.
(D)[13:48:50]{interface.c:230}	Emitting 'BlPct' property
(D)[13:48:51]{backlight.c:736}	Backlight 'intel_backlight' level updated: 1.00.
(D)[13:48:51]{interface.c:230}	Emitting 'BlPct' property
(D)[13:49:12]{timer.c:32}	Set timeout of 30s 0ns on fd 43.
(D)[13:49:12]{backlight.c:514}	Content calib: wmax: 1.181, wmin: 0.781, new_bl: 1.000
(I)[13:49:12]{backlight.c:522}	Ambient brightness: 0.906, Screen brightness: 0.137 -> Screen backlight: 1.000.
(D)[13:49:12]{interface.c:230}	Emitting 'ScreenBr' property
(D)[13:49:12]{interface.c:230}	Emitting 'BlPct' property
(D)[13:49:12]{backlight.c:736}	Backlight 'intel_backlight' level updated: 1.00.
(D)[13:49:12]{interface.c:230}	Emitting 'BlPct' property
(D)[13:49:12]{backlight.c:736}	Backlight 'intel_backlight' level updated: 0.95.
(D)[13:49:12]{interface.c:230}	Emitting 'BlPct' property
```

sonofevil, 2024-03-12:
Well this is odd. It seems that my settings in `/etc/` now work after I deleted `~/.config/clight.conf`, even though it didn't work before, when that file did not exist. All I have changed is that I added regression points from `-w` to `/etc/clight/modules.conf.d/sensor.conf`, although if I comment that out, nothing changes.

I have no explanation for what happened, but the issue seems to be gone. Closing.

closed 2024-03-12 by sonofevil

FedeDP, 2024-03-12:
Are you sure that a local config was not present before? Local config files override global ones, thus that would explain everything!

FedeDP, 2024-03-12:
Anyway, glad you solved the issue! :rocket:

sonofevil, 2024-03-12:
> Because
> 
> > (D)[13:48:12]{config.c:413}	Reading config file: /home/lutra/.config/clight.conf
> 
> That's a debug message only enabled in verbose mode :)

```
(D)[14:55:31]{config.c:413}	Reading config file: /home/lutra/.config/clight.conf
(W)[14:55:31]{opts.c:438}	DPMS_CONF: wrong AC 'timeout' value (<= dimmer timeout). Resetting default value.
(W)[14:55:31]{opts.c:443}	DPMS_CONF: wrong BATT 'timeout' value (<= dimmer timeout). Resetting default value.

Clight
* Software version:		4.12-e256faa
* Global config dir:		/etc/clight
* Global data dir:		/usr/share/clight
* Starting time:		Tue Mar 12 14:55:31 2024

Starting options:

### GENERIC ###
* Verbose (debug):		Disabled
* ResumeDelay:		0
```

sonofevil, 2024-03-12:
No, you were right, the log entry about which file is being read is based on the verbosity set in `/etc/clight/clight.conf`, but then the local file overrides that verbosity.