FedeDP/Clight #297: [BUG] misformatted version string when installing from the release tarballs

ferki, 2024-05-03:
**Clight version the issue has been seen with:**
> 4.11 (but apparently present since 4.8)

**Used distribution:**
> Gentoo

**Describe the bug**
When unpacking the release tarballs to build Clight, the extracted content is not a git repository, therefore cmake complains about "fatal: not a git repository" when trying to determing the git commit hash.

It results in an empty `GIT_HASH` variable, causing the version string to miss the trailing git commit id, and look like `4.11-`

**Expected behavior**
I'd like Clight to have a clean build process without errors/warnings, and to report properly formatted version strings regardless of how one obtains the source code (from git or from release tarballs).

**To Reproduce**
1. Download and extract the release tarball.
2. Run `cmake .` in the extracted source directory.
3. Observe a line starting with `fatal: not a git repository` in the output.
4. Build Clight fully, and run `clight -v` to get the version string.
5. Observe the missing trailing git commit hash after the `-` character.

Given the build-time issue, I decided not to attach a verbose Clight log from runtime for now. It contains the same version string though, so let me know if that's relevant after all.

Additional notes:

1. I currently use a [quick patch](https://github.com/ferki/gentoo-overlay/blob/7f2329c08df56782615636843a57efa2a045ea03/app-misc/clight/files/clight-version.patch) around it in my Gentoo overlay. If that is an acceptable solution for this repo, I would be more than happy to sent it in a PR here.
2. The same issue is apparently also present in Clightd, and I have a similar patch for it too. I can open a similar issue and/or PR in the Clightd repository accordingly, I just wished to get a first feedback early, before making more noise about this :) Let me know how you prefer to proceed!

FedeDP, 2024-05-07:
Hi! Thanks for reporting this! I agree, this is a bad behavior that should be fixed.
> I currently use a [quick patch](https://github.com/ferki/gentoo-overlay/blob/7f2329c08df56782615636843a57efa2a045ea03/app-misc/clight/files/clight-version.patch) around it in my Gentoo overlay. If that is an acceptable solution for this repo, I would be more than happy to sent it in a PR here.

It seems a suited solution! We could also capture the result variable and check if it is not 0, like:
```
execute_process(COMMAND git log -1 --format=%h
WORKING_DIRECTORY ${CMAKE_CURRENT_LIST_DIR}
         OUTPUT_VARIABLE GIT_HASH
         OUTPUT_STRIP_TRAILING_WHITESPACE
RESULT_VARIABLE ret)
if(ret EQUAL "0")
    set(VERSION "${PROJECT_VERSION}-${GIT_HASH}")
else()
   set(VERSION "${PROJECT_VERSION}")
endif()
```

(Also, it would be great if you could open same PR on clightd too, thanks!)

ferki, 2024-05-11:
> It seems a suited solution! We could also capture the result variable and check if it is not 0

Using the git command itself makes it a build-time dependency, which I'd also like to avoid when using source tarballs to install instead of the git tree.

I'm afraid using the exit code would lead to a different error during build if the source tarballs are used _and_ git is not installed on the system.

That's why I ended up with an approach which uses something else than git to check if the directory looks like a git repo (such as `test -d .git` as first approximation).

> (Also, it would be great if you could open same PR on clightd too, thanks!)

Great, thanks, will do both PRs shortly!

closed 2024-05-12 by FedeDP

ferki, 2024-05-12:
I'm glad it was acceptable in both places, thank again for mergin!

PR #298 (2024-05-12): Fix version string formatting
This is an attempt to fix #297 by using git to retrieve the git commit hash only if the source directory looks like a git repository.

Please review and merge, or let me know how to improve it.