FedeDP/Clight #306: [BUG] MISSING DEPENDENCIES

Repressed8077, 2024-07-16:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> 4.11

**Used distribution:**
> Linux Mint 21.3 

**Describe the bug**
The following dependencies lead to clight startup error when not installed: libjansson-dev and libiio-dev

**Expected behavior**
Include the above missing deps under Ubuntu/Linux Mint, etc in the README installation guide.

**To Reproduce**
Use the script provided in https://github.com/FedeDP/Clight/wiki/Build under **Dealing with dependencies**. After the script has completed running clight in the command leads to clightd dep for Clight not found.

Repressed8077, 2024-08-11:
The libiio-dev dependency seems generic because when I was building Clight on Fedora Workstation Edition, the build failed due to this missing dependency.

erflaber, 2024-10-09:
I can cofirm. The build instructions for Ubuntu https://github.com/FedeDP/Clight/wiki/Build#how-to-build should include libiio-dev